NorthernMan54/homebridge-dht #3: homebridge throws error when trying to read temps

CaptBeard, 2017-01-13:
I'm getting the following error when running homebridge. This seems to only trigger when the sensor is being read by homekit.
I have tested my sensor using python so I know it's working correctly.

`[1/12/2017, 8:14:54 PM] Homebridge is running on port 51826.
internal/child_process.js:313
    throw errnoException(err, 'spawn');
    ^

Error: spawn EACCES
    at exports._errnoException (util.js:1022:11)
    at ChildProcess.spawn (internal/child_process.js:313:11)
    at exports.spawn (child_process.js:380:9)
    at exports.execFile (child_process.js:143:15)
    at Object.getDHTTemperature (/usr/lib/node_modules/homebridge-dht/index.js:66:9)
    at emitThree (events.js:116:13)
    at emit (events.js:194:7)
    at Characteristic.getValue (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Characteristic.js:120:10)
    at Bridge.<anonymous> (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Accessory.js:634:20)
    at Array.forEach (native)
    at Bridge.Accessory._handleGetCharacteristics (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Accessory.js:598:8)
    at emitMany (events.js:127:13)
    at HAPServer.emit (events.js:201:7)
    at HAPServer._handleCharacteristics (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/HAPServer.js:926:10)
    at HAPServer.<anonymous> (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/HAPServer.js:209:39)
    at emitNone (events.js:86:13)
    at IncomingMessage.emit (events.js:185:7)
    at endReadableNT (_stream_readable.js:974:12)
    at _combinedTickCallback (internal/process/next_tick.js:74:11)
    at process._tickCallback (internal/process/next_tick.js:98:9)`

Thanks and great work!

NorthernMan54, 2017-01-13:
> at Bridge. (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Accessory.js:634:20)
> at HAPServer. (/usr/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/HAPServer.js:209:39)

Just looking at the error message, and it indicates an access denied error for the dht22 command.

I believe if you try the command

dht22

from the command line, when you try this as the same user that is running homebridge it will give an error.

Change to the directory where you the the dht22 command, which s/b /usr/local/bin and execute

sudo chmod a+rx dht22

then try the command dht22 again, and you shouldn’t receive an error any more.

CaptBeard, 2017-01-13:
dht22 command not recognized.
I followed your readme as best I could

because I installed homebridge globally, I installed dht22 to /usr/bin/dht22
that directory looks like this:
```
/usr/bin/dht22 $ ls -l
total 4
drwxrwxrwx 2 root root 4096 Jan 12 19:53 DHTXXD
```

NorthernMan54, 2017-01-13:
I should change that readme to point at the instructable I created which offers a lot better steps.  Take a look at this, and in particular step 7.

http://www.instructables.com/id/RPIHomeBridge-TemperatureHumidity-Sensor/

CaptBeard, 2017-01-13:
> http://www.instructables.com/id/RPIHomeBridge-TemperatureHumidity-Sensor/

That helped tremendously, thank you, it's working now!

NorthernMan54, 2017-01-13:
Tks

closed 2017-01-13 by NorthernMan54