NorthernMan54/homebridge-dht #5: having error reading the sensor

toshibochan, 2017-05-18:
"accessories": [{
                        "accessory": "Dht",
                        "name": "dht22 - outdoor",
                        "name_temperature": "Outdoor Temperature",
                        "name_humidity": "Outdoor Humdity",
                        "gpio": "26",
                        "service": "dht22"
        }]
}

[5/18/2017, 11:38:52 AM] [dht22 - outdoor] Identify requested!
[5/18/2017, 11:39:13 AM] [dht22 - outdoor] dhtExec function failed: Error: spawn dht22 ENOENT
[5/18/2017, 11:39:13 AM] [dht22 - outdoor] Identify requested!
[5/18/2017, 11:39:16 AM] [dht22 - outdoor] dhtExec function failed: Error: spawn dht22 ENOENT
[5/18/2017, 11:39:17 AM] [dht22 - outdoor] dhtExec function failed: Error: spawn dht22 ENOENT
[5/18/2017, 11:39:23 AM] [dht22 - outdoor] dhtExec function failed: Error: spawn dht22 ENOENT

NorthernMan54, 2017-05-18:
ENONET means file not found. It is looking for a binary called dht22, that needs to be built. The build instructions are part of the README.

toshibochan, 2017-05-18:
i install the DHTXXD.zip
and now give me
"pi@toshiboPI32:~ $ dht22
ERROR: pigpiod not running"

toshibochan, 2017-05-18:
just give me 
ERROR: pigpiod not running

toshibochan, 2017-05-18:
after i run "sudo pigpiod"  work

toshibochan, 2017-05-18:
problem is i have to run "sudo pigpiod" after restart raspberry pi. i have to add on "crontab" to run "sudo pigpiod" on restart.

NorthernMan54, 2017-05-18:
You can add pigpiod to /etc/rc.local

And that will autostart it

closed 2017-07-13 by NorthernMan54