NorthernMan54/homebridge-dht #10: Recent homebridge update broken things?

SkyJohn, 2017-08-16:
The recent update to home bridge (0.4.25) seems to have broken this, homebridge wont start with it installed.

pdlove, 2017-08-16:
As a note over here... Some plugins had issue because Apple changed some characteristics on services with the IOS 11 releases. I've fixed HAP-NodeJS to simply give warnings when these are encountered. For the time being Apple isn't rejecting them, but you never know when they might start. Let me know if you need more information on how to resolve this.

NorthernMan54, 2017-08-16:
@pdlove

Tks for the warning on the changes, I should be okay with making the fixes for this, but I will let you know if I need an assist.  I have been watching these changes for IOS 11 closely, as I maintain a forked version of hap-nodesjs that supports Alexa.   And I expect some challenges with accessory numbering as a result of your

NorthernMan54, 2017-08-16:
@pdlove

Updating via mobile, and it went and closed it

Changes to aid and iid.

pdlove, 2017-08-16:
The changes I made are only to the iids on the bridge and only when there is more than one service. It shouldn’t cause you any issues.

NorthernMan54, 2017-08-16:
I found the issue, I was setting the serial number to "undefined".  It has been fixed and promoted.

SkyJohn, 2017-08-17:
Works now :)

Thanks for the quick update.

closed 2017-10-09 by NorthernMan54