NorthernMan54/homebridge-dht #11: DH11 showing humidity value of NaN

timothy-gibson, 2017-09-23:
I installed this plugin and the temperature of my DH11 sensor works fine but I have not yet been able to get the humidity to show through to homekit. 

Homekit status:

```
Sep 23 04:16:33 mediapi homebridge[5619]: [Sat Sep 23 2017 04:16:33 GMT+0000 (UTC)] [dht22] Got status of 0
Sep 23 04:16:33 mediapi homebridge[5619]: [Sat Sep 23 2017 04:16:33 GMT+0000 (UTC)] [dht22] Got Temperature of 26
Sep 23 04:16:33 mediapi homebridge[5619]: [Sat Sep 23 2017 04:16:33 GMT+0000 (UTC)] [dht22] Got humidity of NaN
```

And if I run the dhtxx binary I get:

```
~ $ dht22
0 26.0 52.0
```

Which shows a humidity of 52%. Any idea why I'm not getting this data flowing through to the plugin?

NorthernMan54, 2017-09-23:
I had patched the dhtxx program as part of the installation, and it appears that this was bypassed.  Trying patching the dhtxx program.   After patching it should work.  Patching detail are included in the instructions.

timothy-gibson, 2017-09-24:
My apologies, it appears I must have missed the patching instructions. Working perfectly now, cheers.

closed 2017-09-24 by timothy-gibson