NorthernMan54/homebridge-dht #12: Can't find the accessory

cequn, 2017-10-09:
Hi, I follow your instruction to set up everything, and when i run the homebridge, all goes well, no error
but the homekit can't pair this accessory,  I also use another plugin called homebridge-camera-rpi, which can be searched by home app. 
so how to figure this out?

cequn, 2017-10-09:
@NorthernMan54  
check the following: 
*** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
*** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
[2017-10-9 23:33:09] Loaded plugin: homebridge-camera-rpi
[2017-10-9 23:33:09] Registering platform 'homebridge-camera-rpi.rpi-camera'
[2017-10-9 23:33:09] ---
[2017-10-9 23:33:11] Loaded plugin: homebridge-dht
[2017-10-9 23:33:11] Registering accessory 'homebridge-dht.Dht'
[2017-10-9 23:33:11] ---
[2017-10-9 23:33:11] Loaded plugin: homebridge-gpio-wpi2
[2017-10-9 23:33:11] Registering platform 'homebridge-gpio-wpi2.WiringPiPlatform'
[2017-10-9 23:33:11] ---
[2017-10-9 23:33:11] Loaded config.json with 1 accessories and 1 platforms.
[2017-10-9 23:33:11] ---
[2017-10-9 23:33:11] Loading 1 platforms...
[2017-10-9 23:33:11] [rpi-camera] Initializing rpi-camera platform...
[2017-10-9 23:33:11] Loading 1 accessories...
[2017-10-9 23:33:11] [Temp/Humidity Sensor] Initializing Dht accessory...
[2017-10-9 23:33:11] [Temp/Humidity Sensor] Adding Accessory
[2017-10-9 23:33:11] [Temp/Humidity Sensor] INIT: Temp/Humidity Sensor
[2017-10-9 23:33:11] [Temp/Humidity Sensor] call this getServices
Scan this code with your HomeKit App on your iOS device to pair with Homebridge:
                       
    ┌────────────┐     
    │ 031-45-154 │     
    └────────────┘     
                       
[2017-10-9 23:33:11] Homebridge is running on port 51826.
[2017-10-9 23:33:11] Pi Camera is running on port 40517.

NorthernMan54, 2017-10-09:
Its added correctly, do you look in the “Default Room” of your home?   Once you have paired homebridge to your phone, any added accessories end up in the default room.

cequn, 2017-10-10:
@NorthernMan54 there is no any other accessory on my homekit,  but is it possible that the ios 11.0.2, the latest version, doesn't support?

cequn, 2017-10-10:
@NorthernMan54 Thank you. I figure it out.

closed 2017-10-10 by NorthernMan54