NorthernMan54/homebridge-dht #17: Support for dht11

sincetomorrow, 2018-03-16:
Hello and nice work!! 
One question, does it support dht11? I have tried to replay at the service dht11 instead of dht22 but had no luck.

Thank you

NorthernMan54, 2018-03-16:
Panos,

It should but I have don’t have one to test, so it may be a little bit of trial and error.

The core of the plugin is the dht22 binary, that gets run every time the plugin wants temperature.  If you run the dht22 command does it return temp and humidity on the command line?

If yes, the don’t change anything from the plugin config.

Namelessinterester, 2018-04-14:
It works fine with my DHT11, but "service":   "dht22", so I just removed it from my config.

LeoCal, 2018-06-08:
Great job @NorthernMan54 !

NorDroN, 2018-06-14:
@Namelessinterester, could you give a part of your config for dht11?

NorthernMan54, 2018-06-15:
The config for DHT11 and DHT22 is the same, just follow the directions for the dht22, and don’t change anything.  It should just work.

NorDroN, 2018-06-15:
@NorthernMan54 unfortunately it does't work, I see 0°C and 0% :(

NorthernMan54, 2018-06-15:
And if you run the dht22 command from the command line do you see the same?

closed 2018-11-08 by NorthernMan54