NorthernMan54/homebridge-dht #18: Not able to log temp/hum records

mruwzum, 2018-09-08:
Even if I try to set it to "fs" and look for it in /var/homebride/persist/ or choose another directory, I'm not able to generate any kind of JSON record. Would like to use it with EVE app.

NorthernMan54, 2018-09-08:
The plugin defaults to FS, so you don’t have to do anything in the config file, have you tried the EVE App?

mruwzum, 2018-09-08:
Tried many times without results, and even no JSON found.

mruwzum, 2018-09-08:
[image]

NorthernMan54, 2018-09-08:
Humm, it should just work, can you share some Logs from homebridge, I think it is having issues

mruwzum, 2018-09-08:
It doesn't report any kind of warn or error, just the temp/humidity each minute. Tried reinstalling, but no changes.

There's a recent log from Homebridge.

Started Node.js HomeKit Server.
pam_unix(sudo:session): session closed for user root
[2018-9-8 20:00:18] Loaded config.json with 1 accessories and 2 platforms.
[2018-9-8 20:00:18] ---
[2018-9-8 20:00:18] Loaded plugin: homebridge-config-ui-x
[2018-9-8 20:00:18] Registering platform 'homebridge-config-ui-x.config'
[2018-9-8 20:00:18] ---
[2018-9-8 20:00:19] Loaded plugin: homebridge-dht
[2018-9-8 20:00:19] Registering accessory 'homebridge-dht.Dht'
[2018-9-8 20:00:19] ---
[2018-9-8 20:00:19] Loaded plugin: homebridge-gpio-wpi2
[2018-9-8 20:00:19] Registering platform 'homebridge-gpio-wpi2.WiringPiPlatform'
[2018-9-8 20:00:19] ---
[2018-9-8 20:00:19] Loading 2 platforms...
[2018-9-8 20:00:19] [Config] Initializing config platform...
[2018-9-8 20:00:19] [Config] Spawning homebridge-config-ui-x with PID 1533
[2018-9-8 20:00:19] [Pi GPIO (WiringPi)] Initializing WiringPiPlatform platform...
[2018-9-8 20:00:19] [Pi GPIO (WiringPi)] WORK IN PROGRESS... Report issues on https://github.com/rsg98/homebridge-gpio-wpi2
[2018-9-8 20:00:19] [Pi GPIO (WiringPi)] Exporting GPIO Pins from config file...
[2018-9-8 20:00:19] [Pi GPIO (WiringPi)] Exporting and configuring: GPIO2
[2018-9-8 20:00:19] [Pi GPIO (WiringPi)] Exporting and configuring: GPIO3
[2018-9-8 20:00:19] [Pi GPIO (WiringPi)] homebridge API version: 2.2
[2018-9-8 20:00:19] Loading 1 accessories...
[2018-9-8 20:00:19] [Temp/Humidity Sensor] Initializing Dht accessory...
[2018-9-8 20:00:19] [Temp/Humidity Sensor] Adding Accessory
[2018-9-8 20:00:19] [Temp/Humidity Sensor] INIT: Temp/Humidity Sensor
[2018-9-8 20:00:19] [Pi GPIO (WiringPi)] GPIO2 Configure GPIO Pin 9aec2d2c-b026-4884-8b57-8b45093a8559
[2018-9-8 20:00:19] [Pi GPIO (WiringPi)] Turning off pin #23
[2018-9-8 20:00:19] [Pi GPIO (WiringPi)] GPIO3 Configure GPIO Pin 6e9aeda2-0969-4a0b-9367-e53406d6d86a
[2018-9-8 20:00:19] [Pi GPIO (WiringPi)] Turning off pin #18
Setup Payload:
X-HM://0024W8RSEE22G
Scan this code with your HomeKit app on your iOS device to pair with Homebridge:
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
                                                      
Or enter this code with your HomeKit app on your iOS device to pair with Homebridge:
                       
    ┌────────────┐
    │ 861-83-598 │
    └────────────┘
                       
[... 40 lines cut ...]
[2018-9-8 20:31:19] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.3, Humidity: 54.6
[2018-9-8 20:32:19] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.3, Humidity: 54.7
[2018-9-8 20:33:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.3, Humidity: 54.7
[2018-9-8 20:34:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.2, Humidity: 54.8
[2018-9-8 20:35:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.2, Humidity: 54.9
[2018-9-8 20:36:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.2, Humidity: 54.9
[2018-9-8 20:37:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.3, Humidity: 55
[2018-9-8 20:38:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.2, Humidity: 55.2
[2018-9-8 20:39:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.3, Humidity: 55.5
[2018-9-8 20:40:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.3, Humidity: 55.7
[2018-9-8 20:41:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.3, Humidity: 55.9
[2018-9-8 20:42:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.3, Humidity: 56
[2018-9-8 20:43:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.2, Humidity: 56.2
[2018-9-8 20:44:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.3, Humidity: 56.3
[2018-9-8 20:45:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.3, Humidity: 56.3
[2018-9-8 20:46:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.2, Humidity: 56.4
[2018-9-8 20:47:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.3, Humidity: 56.4
[... cut ...]
[2018-9-8 20:49:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.4, Humidity: 56.6
[2018-9-8 20:50:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.3, Humidity: 56.3
[2018-9-8 20:50:45] [Config] Changes to config.json saved.
[2018-9-8 20:51:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.2, Humidity: 56
[2018-9-8 20:52:20] [Temp/Humidity Sensor] DHT Status: 0, Temperature: 26.2, Humidity: 56.4
[2018-9-8 20:52:45] [Config] Changes to config.json saved.
[2018-9-8 20:52:46] [Config] Homebridge restart request received

NorthernMan54, 2018-09-08:
Can you try a debug log?

DEBUG=* homebridge

mruwzum, 2018-09-08:
It says that adds data, but I can't find it. 

pi@raspb3:~ $ homebridge -D
[2018-9-8 22:01:53] Loaded config.json with 1 accessories and 2 platforms.
[2018-9-8 22:01:54] ---
[2018-9-8 22:01:54] Loaded plugin: homebridge-config-ui-x
[2018-9-8 22:01:54] Registering platform 'homebridge-config-ui-x.config'
[2018-9-8 22:01:54] ---
[2018-9-8 22:01:55] Loaded plugin: homebridge-dht
[2018-9-8 22:01:55] Registering accessory 'homebridge-dht.Dht'
[2018-9-8 22:01:55] ---
[2018-9-8 22:01:55] Loaded plugin: homebridge-gpio-wpi2
[2018-9-8 22:01:55] Registering platform 'homebridge-gpio-wpi2.WiringPiPlatform'
[2018-9-8 22:01:55] ---
[2018-9-8 22:01:55] Loading 2 platforms...
[2018-9-8 22:01:55] [Config] Initializing config platform...
[2018-9-8 22:01:55] [Config] Spawning homebridge-config-ui-x with PID 1730
[2018-9-8 22:01:55] [Pi GPIO (WiringPi)] Initializing WiringPiPlatform platform...
[2018-9-8 22:01:55] [Pi GPIO (WiringPi)] WORK IN PROGRESS... Report issues on https://github.com/rsg98/homebridge-gpio-wpi2
[2018-9-8 22:01:55] [Pi GPIO (WiringPi)] Exporting GPIO Pins from config file...
[2018-9-8 22:01:55] [Pi GPIO (WiringPi)] Exporting and configuring: GPIO2
[2018-9-8 22:01:55] [Pi GPIO (WiringPi)] Exporting and configuring: GPIO3
[2018-9-8 22:01:55] [Pi GPIO (WiringPi)] homebridge API version: 2.2
[2018-9-8 22:01:55] Loading 1 accessories...
[2018-9-8 22:01:55] [dht22] Initializing Dht accessory...
[2018-9-8 22:01:55] [dht22] Adding Accessory
[2018-9-8 22:01:55] [dht22] INIT: dht22
[2018-9-8 22:01:55] [dht22] ** Fakegato-storage AddWriter : Temperature
[2018-9-8 22:01:55] [dht22] Loading...
[2018-9-8 22:01:55] [dht22] ** Fakegato-storage read FS file: /home/pi/.homebridge/raspb3_Temperature_persist.json
[2018-9-8 22:01:55] [dht22] ** Fakegato-timer Subscription : Temperature
[2018-9-8 22:01:55] [dht22] Registring Events Temperature
[2018-9-8 22:01:55] [dht22] Accessory Temperature
[2018-9-8 22:01:55] [Pi GPIO (WiringPi)] GPIO2 Configure GPIO Pin 9aec2d2c-b026-4884-8b57-8b45093a8559
[2018-9-8 22:01:55] [Pi GPIO (WiringPi)] Turning off pin #23
[2018-9-8 22:01:55] [Pi GPIO (WiringPi)] GPIO3 Configure GPIO Pin 6e9aeda2-0969-4a0b-9367-e53406d6d86a
[2018-9-8 22:01:55] [Pi GPIO (WiringPi)] Turning off pin #18
Setup Payload:
X-HM://0024W8RSECNJ7
Scan this code with your HomeKit app on your iOS device to pair with Homebridge:

[2018-9-8 22:01:55] [Pi GPIO (WiringPi)] Loading cached GPIO pins complete
[2018-9-8 22:01:55] Homebridge is running on port 54233.
[2018-9-8 22:01:55] [dht22] DHT Status: 0, Temperature: 25.1, Humidity: 51.4
[2018-9-8 22:01:55] [dht22] **Fakegato-timer: addData  Temperature { time: 1536444115, temp: 25.1, humidity: 51.4 }  immediate:  false
[2018-9-8 22:01:55] [dht22] **Start Global Fakegato-Timer - 10min**
[2018-9-8 22:01:59] [Config] Using Form Authentication
[2018-9-8 22:01:59] [Config] Console v3.8.10 is listening on port 8080.
[2018-9-8 22:02:55] [dht22] DHT Status: 0, Temperature: 25.1, Humidity: 51.4
[2018-9-8 22:02:55] [dht22] **Fakegato-timer: addData  Temperature { time: 1536444175, temp: 25.1, humidity: 51.4 }  immediate:  false
[2018-9-8 22:03:26] [dht22] Identify requested!
[2018-9-8 22:03:32] [dht22] Identify requested!
[2018-9-8 22:03:41] [Pi GPIO (WiringPi)] GPIO2 Identify!!!
[2018-9-8 22:03:54] [Pi GPIO (WiringPi)] GPIO3 Identify!!!
[2018-9-8 22:03:55] [dht22] DHT Status: 0, Temperature: 25, Humidity: 51.2
[2018-9-8 22:03:55] [dht22] **Fakegato-timer: addData  Temperature { time: 1536444235, temp: 25, humidity: 51.2 }  immediate:  false
[2018-9-8 22:04:53] [Pi GPIO (WiringPi)] Turning on pin #23
[2018-9-8 22:04:54] [Pi GPIO (WiringPi)] Turning off pin #23
[2018-9-8 22:04:54] [Pi GPIO (WiringPi)] Turning on pin #23
[2018-9-8 22:04:55] [dht22] DHT Status: 0, Temperature: 25, Humidity: 51.3
[2018-9-8 22:04:55] [dht22] **Fakegato-timer: addData  Temperature { time: 1536444295, temp: 25, humidity: 51.3 }  immediate:  false
[2018-9-8 22:04:55] [Pi GPIO (WiringPi)] Turning off pin #23
[2018-9-8 22:04:55] [Pi GPIO (WiringPi)] Turning on pin #18
[2018-9-8 22:05:55] [dht22] DHT Status: 0, Temperature: 24.9, Humidity: 51.4
[2018-9-8 22:05:55] [dht22] **Fakegato-timer: addData  Temperature { time: 1536444355, temp: 24.9, humidity: 51.4 }  immediate:  false

---------------------------

The file raspb3_Temperature_persist.json doesn't exist on /home/pi/.homebridge/

NorthernMan54, 2018-09-08:
Is there a permissions issue?  Does the account your running homebridge have the ability to write to that directory?

mruwzum, 2018-09-09:
When starting it like "homebridge -D" it's able to record the logs, but without the D, it doesn't save any record.

[2018-9-9 10:46:52] [dht22] Clock adjust Temperature: 9c314521
[2018-9-9 10:47:23] [dht22] DHT Status: 3, Temperature: 0, Humidity: 0
[2018-9-9 10:48:12] [dht22] **Fakegato-timer: executeCallbacks**
[2018-9-9 10:48:12] [dht22] First entry Temperature: 0
[2018-9-9 10:48:12] [dht22] Last entry Temperature: 49
[2018-9-9 10:48:12] [dht22] Used memory Temperature: 49
[2018-9-9 10:48:12] [dht22] 116 Temperature: 41b1000000000000ab80442103 0102 0202 03024a00c00f00000000000000000101
[2018-9-9 10:48:12] [dht22] ** Fakegato-storage write FS file: /home/pi/.homebridge/raspb3_Temperature_persist.json "firstEntry":0,"lastEntry":73,"usedMemory":73,"refTime":558137515,"initialTime":
[2018-9-9 10:48:12] [dht22] **Fakegato-timer: emptyData ** Temperature
[2018-9-9 10:48:23] [dht22] DHT Status: 0, Temperature: 26.7, Humidity: 56
[2018-9-9 10:48:23] [dht22] **Fakegato-timer: addData  Temperature { time: 1536490103, temp: 26.7, humidity: 56 }  immediate:  false
[2018-9-9 10:49:23] [dht22] DHT Status: 0, Temperature: 26.7, Humidity: 56
[2018-9-9 10:49:23] [dht22] **Fakegato-timer: addData  Temperature { time: 1536490163, temp: 26.7, humidity: 56 }  immediate:  false
[2018-9-9 10:50:23] [dht22] DHT Status: 0, Temperature: 26.7, Humidity: 56
[2018-9-9 10:50:23] [dht22] **Fakegato-timer: addData  Temperature { time: 1536490223, temp: 26.7, humidity: 56 }  immediate:  false
[2018-9-9 10:51:23] [dht22] DHT Status: 0, Temperature: 26.7, Humidity: 55.9
[2018-9-9 10:51:23] [dht22] **Fakegato-timer: addData  Temperature { time: 1536490283, temp: 26.7, humidity: 55.9 }  immediate:  false
[2018-9-9 10:52:23] [dht22] DHT Status: 0, Temperature: 26.7, Humidity: 56.1
[2018-9-9 10:52:23] [dht22] **Fakegato-timer: addData  Temperature { time: 1536490343, temp: 26.7, humidity: 56.1 }  immediate:  false
[2018-9-9 10:53:23] [dht22] DHT Status: 0, Temperature: 26.7, Humidity: 56
[2018-9-9 10:53:23] [dht22] **Fakegato-timer: addData  Temperature { time: 1536490403, temp: 26.7, humidity: 56 }  immediate:  false
[2018-9-9 10:54:23] [dht22] DHT Status: 3, Temperature: 0, Humidity: 0
[2018-9-9 10:55:23] [dht22] DHT Status: 0, Temperature: 26.7, Humidity: 55.8
[2018-9-9 10:55:23] [dht22] **Fakegato-timer: addData  Temperature { time: 1536490523, temp: 26.7, humidity: 55.8 }  immediate:  false
[2018-9-9 10:56:23] [dht22] DHT Status: 0, Temperature: 26.7, Humidity: 55.7
[2018-9-9 10:56:23] [dht22] **Fakegato-timer: addData  Temperature { time: 1536490583, temp: 26.7, humidity: 55.7 }  immediate:  false
[2018-9-9 10:57:23] [dht22] DHT Status: 0, Temperature: 26.7, Humidity: 55.8
[2018-9-9 10:57:23] [dht22] **Fakegato-timer: addData  Temperature { time: 1536490643, temp: 26.7, humidity: 55.8 }  immediate:  false
[2018-9-9 10:58:12] [dht22] **Fakegato-timer: executeCallbacks**
[2018-9-9 10:58:12] [dht22] First entry Temperature: 0
[2018-9-9 10:58:12] [dht22] Last entry Temperature: 4a
[2018-9-9 10:58:12] [dht22] Used memory Temperature: 4a
[2018-9-9 10:58:12] [dht22] 116 Temperature: 99b3000000000000ab80442103 0102 0202 03024b00c00f00000000000000000101
[2018-9-9 10:58:12] [dht22] ** Fakegato-storage write FS file: /home/pi/.homebridge/raspb3_Temperature_persist.json "firstEntry":0,"lastEntry":74,"usedMemory":74,"refTime":558137515,"initialTime":
[2018-9-9 10:58:12] [dht22] **Fakegato-timer: emptyData ** Temperature
[2018-9-9 10:58:23] [dht22] DHT Status: 0, Temperature: 26.8, Humidity: 55.8
[2018-9-9 10:58:23] [dht22] **Fakegato-timer: addData  Temperature { time: 1536490703, temp: 26.8, humidity: 55.8 }  immediate:  false
[2018-9-9 10:59:23] [dht22] DHT Status: 0, Temperature: 26.8, Humidity: 55.8
[2018-9-9 10:59:23] [dht22] **Fakegato-timer: addData  Temperature { time: 1536490763, temp: 26.8, humidity: 55.8 }  immediate:  false
[2018-9-9 10:59:52] [Pi GPIO (WiringPi)] Turning on pin #23
[2018-9-9 10:59:52] [Pi GPIO (WiringPi)] Turning on pin #18
[2018-9-9 10:59:52] [Pi GPIO (WiringPi)] Turning off pin #23
[2018-9-9 10:59:52] [Pi GPIO (WiringPi)] Turning on pin #18

NorthernMan54, 2018-09-09:
I’m out of the country for a few weeks, and don’t have access to my computer to take a deeper look at this, can we hold off until I’m back home?

mruwzum, 2018-09-09:
No problem. Meanwhile I'm running it in debug mode so I can log temps, thanks!

NorthernMan54, 2018-09-09:
I run mine in debug all the time, as i like the logs for troubleshooting

closed 2019-01-25 by NorthernMan54