NorthernMan54/homebridge-dht #20: Can't start plugin on boot

akaminetzkyp, 2019-02-03:
Hello, @NorthernMan54!

First of all, thank you very much for this plugin. It's been really useful.

I tested `homebridge-dht` by running `sudo pigpiod` and then `homebridge`. That worked perfectly. My problems began when I tried to run homebridge on boot. I configured cron to run the two commands mentioned before, but this didn't work. I'm not really sure why, but maybe `sudo pigpiod` wasn't being called before `homebridge`. Afterwards, I tried with `sudo systemctl enable pigpiod` and a script that called `homebridge` after waiting 15 seconds with cron. This method didn't work, too.

Homebridge starts successfully with these methods, but the DHT plugin does not. If my problem is that `sudo pigpiod` is not called before `homebridge`, how can I make sure it is called first?

Thank you very much in advance!

akaminetzkyp, 2019-02-03:
I think I used the same method. When I run the second command, the following is shown.
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
pigpio is already the newest version (1.64-1).
python-pigpio is already the newest version (1.64-1).
python3-pigpio is already the newest version (1.64-1).
The following packages were automatically installed and are no longer required:
  arduino-core avr-libc avrdude binutils-avr extra-xdg-menus gcc-avr libftdi1
  libjna-java libjna-jni librxtx-java realpath
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
```

Thank you very much for your quick reply!

NorthernMan54, 2019-02-04:
> It appears that pigpiod is running on boot, as this is shown when I call sudo pigpiod.
>
> 2019-02-03 19:22:25 initInitialise: Can't lock /var/run/pigpio.pid
> Can't initialise pigpio library
> I can run homebridge manually, too.

I think I’m a bit lost here, what is the actual problem / issue

If pigpiod starts after homebridge, it’s not a real issue as the next time homebridge looks for the temperature it will work.

akaminetzkyp, 2019-02-04:
I had no idea the plugin worked when pigpiod started after homebridge. Just did that and homebridge got the data the second time.

Thank you so much!

akaminetzkyp, 2019-02-04:
My problem was that the temperature/humidity data in the Home app was showing "no response". When I waited for the second data poll, it worked flawlessly.

closed 2019-02-04 by akaminetzkyp