NorthernMan54/homebridge-dht #21: Error: dht22 read failed with status 2

danio0701, 2019-03-03:
Can you help me figure out what is happening?

I have errors in homebridge
[2019-3-3 21:30:01] [dht22] DHT Status: 2, Temperature: 0, Humidity: 0
[2019-3-3 21:30:01] [dht22] Error: dht22 read failed with status 2

Sometimes temperature is read correctly. I'm sure that DHT11 is working fine as from AdaFruit DHT is read correctly if I will check this sensor using this script. From your script I have correct reply only on 5% of request.

Next:

[2019-3-3 21:31:01] [dht22] Error: dht22 read failed with status 2 <-- this reply is more often, due that temperature is not displayed correctly in Home app.
[2019-3-3 21:32:01] [dht22] DHT Status: 0, Temperature: 26, Humidity: 38

danio0701, 2019-03-03:
> Can you help me figure out what is happening?
>
> I have errors in homebridge
> [2019-3-3 21:30:01] [dht22] DHT Status: 2, Temperature: 0, Humidity: 0
> [2019-3-3 21:30:01] [dht22] Error: dht22 read failed [...]

Now I noticed also error:
Error: dht22 read failed with status 3

NorthernMan54, 2019-03-03:
The plugin is just passing the errors it gets from the binary dht22.  Try running it from the command line several times in a row, it should exhibit similar behaviour.

Do you have a wiring issue maybe?

danio0701, 2019-03-03:
> The plugin is just passing the errors it gets from the binary dht22. Try running it from the command line several times in a row, it should exhibit similar behavior. Do you have a wiring issue maybe?
> […](#)

Do you have a wiring issue maybe? No, when I'm using Adafruit_Python_DHT then I always have temp and hum.

From terminal command DHT22:

2 0.0 C 0.0 %
3 0.0 C 0.0 %
2 0.0 C 0.0 %
3 0.0 C 0.0 %
2 0.0 C 0.0 %
3 0.0 C 0.0 %
2 0.0 C 0.0 %
3 0.0 C 0.0 %
2 0.0 C 0.0 %
3 0.0 C 0.0 %
2 0.0 C 0.0 %
3 0.0 C 0.0 %
2 0.0 C 0.0 %
3 0.0 C 0.0 %
2 0.0 C 0.0 %
3 0.0 C 0.0 %
2 0.0 C 0.0 %

NorthernMan54, 2019-03-04:
Sounds like the gpio pin configuration problem, the codebase assumes a connection to gpio 4.  Where are you connected to?

danio0701, 2019-03-04:
I'm connected to gpio 4 (second pin on bottom)

NorthernMan54, 2019-03-04:
I’m not sure what is happening here as I’m using pigpiod to talk to the dht device and it seems to be having issues intermittently.  The pigpiod library is here

http://abyz.me.uk/rpi/pigpio/index.html

I’m thinking it should be simple to switch to using the adafruit library instead, only problem is that I’m out of the country for a week and don’t have the ability to look at this in detail.  The plugin just runs a command on the command line and parses the result from it.

danio0701, 2019-03-04:
> I’m not sure what is happening here as I’m using pigpiod to talk to the dht device and it seems to be having issues intermittently. The pigpiod library is here [...]

Oki, thanks a lot for suggestions :) I'm new in this so I will try to read more how to switch these libraries.

danio0701, 2019-03-05:
Is there source code of DHT22 file which I'm opening using command DHT22 and any instruction how to make from this code service again? I'm new in this stuff :P

NorthernMan54, 2019-03-06:
Source is here

http://abyz.me.uk/rpi/pigpio/code/DHTXXD.zip

I did add a minor change to the source which is a patch in repo to include the Status and make the response easier to parse.

danio0701, 2019-03-06:
I generated new DHTXXD, only changes I made in test_DHTXXD to select manually that I have DHT11. Now I have this:

pi@raspberrypi:~/DHTXXD-2 $ ./DHTXXD -i1
2 0.0 0.0
3 0.0 0.0
2 0.0 0.0
3 0.0 0.0
2 0.0 0.0
3 0.0 0.0
2 0.0 0.0
3 0.0 0.0
2 0.0 0.0
3 0.0 0.0
2 0.0 0.0
3 0.0 0.0
0 26.0 37.0
3 26.0 37.0
2 26.0 37.0
3 26.0 37.0
0 26.0 35.0
3 26.0 35.0
0 26.0 34.0
3 26.0 34.0
0 26.0 34.0
3 26.0 34.0
0 26.0 33.0
3 26.0 33.0
2 26.0 33.0
3 26.0 33.0
0 26.0 30.0
3 26.0 30.0
0 26.0 29.0
3 26.0 29.0
0 26.0 28.0
3 26.0 28.0
0 26.0 30.0
3 26.0 30.0

Correct read is a show until I stop reading from the sensor. When I start again command ./DHTXXD -1 then again few first reads will be:

2 0.0 0.0
3 0.0 0.0
2 0.0 0.0
3 0.0 0.0
2 0.0 0.0

Sometimes this will not work and reads only
2 0.0 0.0
3 0.0 0.0
2 0.0 0.0
3 0.0 0.0
2 0.0 0.0
...

It's too difficult for me

NorthernMan54, 2019-03-06:
What I was thinking was to change the parsing logic here

https://github.com/NorthernMan54/homebridge-dht/blob/dd0f61ac18399a41c5aa62fcab5ae8460c0e97a9/index.js#L85

To handle the response from the adafruit program, and ditch the dht22 binary going forward.

danio0701, 2019-03-06:
Still I do not understand how to do that but I noticed that if I will add to config.json of homebridge:

{ 
    "accessory": "Dht",
    "name": "Test-DHT",
    "service": "dht22",
    "dhtExec": "/home/pi/Adafruit_Python_DHT/examples/AdafruitDHT.py"
}

Run homebridge then I have something like this:

DHT Status: Temp=25.0*, Temperature: NaN, Humidity: NaN
[2019-3-6 23:17:50] [Test-DHT] Error: dht22 read failed with status Temp=25.0*
[2019-3-6 23:18:56] [Test-DHT] DHT Status: Temp=25.0*, Temperature: NaN, Humidity: NaN
[2019-3-6 23:18:56] [Test-DHT] Error: dht22 read failed with status Temp=25.0*

Temperature is reading correctly. Humidity is not displayed in this error. I will try to change output, maybe then this will be work :)

danio0701, 2019-03-06:
YES!!! It works!
in homebridge config.json I add accessory:

{ 
    "accessory": "Dht",
    "name": "Test-DHT",
    "service": "dht22",
    "dhtExec": "/home/pi/Adafruit_Python_DHT/examples/AdafruitDHT.py"
}

and I changed code of AdafruitDHT.py output format and set this as defualt DHT11:

import sys

import Adafruit_DHT

humidity, temperature = Adafruit_DHT.read_retry(11, 4)
if humidity is not None and temperature is not None:
    print('0 {0:0.1f} C {1:0.1f}%'.format(temperature, humidity))
else:
    print('Failed to get reading. Try again!')
    sys.exit(1)

NorthernMan54, 2019-03-06:
Congrats

danio0701, 2019-03-06:
> Congrats
> […](#)

Thank you for assistance and suggestions ;)

closed 2019-03-06 by danio0701