NorthernMan54/homebridge-dht #22: Segmentation fault

toshibochan, 2019-03-12:
I'm using pin 16. I changed the pin in dht22 file in /user/local/bin/ and run "dht22" but give:
Segmentation fault

please need help

NorthernMan54, 2019-03-12:
Another user recently switch to using the adafruit library instead of the supplied dht22 binary, details are here.

https://github.com/NorthernMan54/homebridge-dht/issues/21#issuecomment-470302826

NorthernMan54, 2019-04-09:
@toshibochan I changed the package and am now using a different set of libraries for the DHT22.  These may work better.

tomtastic, 2019-04-19:
`  CXX(target) Release/obj.target/node_dht_sensor/dht-sensor.o
  SOLINK_MODULE(target) Release/obj.target/node_dht_sensor.node
/usr/bin/ld: /usr/local/lib/libbcm2835.a(bcm2835.o): relocation R_X86_64_PC32 against symbol `bcm2835_peripherals' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: nonrepresentable section on output
collect2: error: ld returned 1 exit status
make: *** [node_dht_sensor.target.mk:133: Release/obj.target/node_dht_sensor.node] Error 1
make: Leaving directory '/usr/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 5.0.0-11-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build"
gyp ERR! cwd /usr/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor
gyp ERR! node -v v10.15.3
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-dht-sensor@0.0.34 postinstall: `node-gyp build`
npm ERR! Exit status 1`

tomtastic, 2019-04-19:
OK, new library needed compiling with -fPIC as so :

`cd bcm2835
./configure CFLAGS=-fPIC CXXFLAGS=-fPIC
make
make check
make install
`

All good now 👍

NorthernMan54, 2019-04-19:
Is there anything special in your setup requiring this parameter?  I have an RPI 3 running Raspbian and didn't need this.

NorthernMan54, 2019-04-19:
Also tks for this, I updated the Build.md doc for others

https://github.com/NorthernMan54/homebridge-dht/blob/master/Build.md#step-7-install-homebridge-dht

tomtastic, 2019-04-19:
> Is there anything special in your setup requiring this parameter? I have an RPI 3 running Raspbian and didn't need this.

Yes, I'm not running on a RPI currently, but use homebridge-dht to call a BASH script (for lm-sensors) for showing my NAS temperature : 
```
"accessory": "Dht",
            "cputemp": "/usr/local/bin/sensors_homebridge",
```

NorthernMan54, 2019-04-19:
Looks like my switch from using the dht22 binary to a node_js module caused you some challenges.

closed 2019-04-19 by NorthernMan54