NorthernMan54/homebridge-dht #26: got an error

alexanderre, 2019-05-01:
Hi there. Great guide and a wonderful idea, but for some reason after getting through it I get the homebridge to run for a few seconds and it stops. After entering troubleshooting on my phone I see the following: 
pi@raspberrypi:/root/homebridge $ sudo homebridge -U /
[2019-5-2 01:38:28] config.json (/config.json) not found.
[2019-5-2 01:38:28] ====================
[2019-5-2 01:38:28] ERROR LOADING PLUGIN homebridge-dht:
[2019-5-2 01:38:28] Error: Cannot find module 'node-dht-sensor'
    at Function.Module._resolveFilename (module.js:543:15)
    at Function.Module._load (module.js:470:25)
    at Module.require (module.js:593:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge-dht/index.js:40:14)
    at Module._compile (module.js:649:30)
    at Object.Module._extensions..js (module.js:660:10)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:501:12)
    at Function.Module._load (module.js:493:3)
[2019-5-2 01:38:28] ====================

alexanderre, 2019-05-01:
solved - sudo npm install homebridge-dht --unsafe-perm got it done right

NorthernMan54, 2019-05-01:
Great news

closed 2019-05-01 by NorthernMan54