NorthernMan54/homebridge-dht #27: about DHT11 get data failed

ryanrs-chang, 2019-05-12:
Hi,

I must modify the read method of  `getDHTTemperature` 

```
sensor.read(11, this.gpio, function(err, temperature, humidity) {
...
```
instead of
```
sensor.read(22, this.gpio, function(err, temperature, humidity) {
...
```

NorthernMan54, 2019-05-12:
Tks for this, I went and made a minor change to the code and it should now look at the service config.json option and use the value as part of the call to sensor.read

So if service is

"service":          "dht22"

It should pass 22 as the sensor type

And if service was

"service":          "dht11"

It should now pass an 11

NorthernMan54, 2019-05-12:
This is with version v0.1.26

iShift, 2019-05-13:
@NorthernMan54  after last commit my DHT22 stop working =( in logs i get:
[2019-5-13 21:27:45] [dht22] Error: Error: sensor type is invalid

Config:
`  {
            "accessory": "Dht",
            "name": "dht22",
            "name_temperature": "Temperature",
            "name_humidity": "Humidity",
            "service": "dht22",
            "refresh": "30"
        },`

NorthernMan54, 2019-05-14:
Please install the latest version, v0.1.27.  Small typo on my part.  For your dht11, please change service to "dht11"

iShift, 2019-05-14:
Thanks, last version help me :) (I just update, don’t change config)

NorthernMan54, 2019-05-15:
Glad to hear your good to go

closed 2019-05-15 by NorthernMan54