NorthernMan54/homebridge-dht #29: Homebridge crashes when trying to initialize Dht

melgu, 2019-05-14:
```
May 14 19:05:59 raspberrypi2 homebridge[717]: [2019-5-14 19:05:59] [dht22] Initializing Dht accessory...
May 14 19:05:59 raspberrypi2 homebridge[717]: [2019-5-14 19:05:59] [dht22] Adding Accessory
May 14 19:06:00 raspberrypi2 homebridge[717]: [2019-5-14 19:06:00] [dht22] INIT: dht22
May 14 19:06:00 raspberrypi2 systemd[1]: homebridge.service: Main process exited, code=killed, status=11/SEGV
May 14 19:06:00 raspberrypi2 systemd[1]: homebridge.service: Unit entered failed state.
May 14 19:06:00 raspberrypi2 systemd[1]: homebridge.service: Failed with result 'signal'.
```

NorthernMan54, 2019-05-14:
The log should have more details around why homebridge exited.   If you could share we should be able to determine why homebridge exited.

rsuard, 2019-05-24:
Hello there,

Thanks for your work ! 

I had a hombridge server working fine with ligthify, I tried to follow your steps, however, 
I have kind of the same problem, error is here :

 Error: listen EADDRINUSE: address already in use :::51826
    at Server.setupListenHandle [as _listen2] (net.js:1270:14)
    at listenInCluster (net.js:1318:12)
    at Server.listen (net.js:1405:7)
    at EventedHTTPServer.listen (/home/pi/.nvm/versions/node/v10.15.3/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/util/eventedhttp.js:60:19)
    at HAPServer.listen (/home/pi/.nvm/versions/node/v10.15.3/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/HAPServer.js:158:20)
    at Bridge.Accessory.publish (/home/pi/.nvm/versions/node/v10.15.3/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Accessory.js:616:16)
    at Server._publish (/home/pi/.nvm/versions/node/v10.15.3/lib/node_modules/homebridge/lib/server.js:132:16)
    at Server.<anonymous> (/home/pi/.nvm/versions/node/v10.15.3/lib/node_modules/homebridge/lib/server.js:421:14)
    at /home/pi/.nvm/versions/node/v10.15.3/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/util/once.js:16:19
    at self.getDevices.then (/home/pi/.nvm/versions/node/v10.15.3/lib/node_modules/homebridge-lightify/index.js:81:7)
[2019-5-24 15:24:55] Got SIGTERM, shutting down Homebridge...
[2019-5-24 15:24:55] [PiTemp] Error: [Error: failed to read sensor]

I tried to unaccessorize the JSON. I have still the error, and the shutting down ....
What do you think ? 

Thanks a lot,
 ++

NorthernMan54, 2019-05-24:
> at Server. (/home/pi/.nvm/versions/node/v10.15.3/lib/node_modules/homebridge/lib/server.js:421:14)

This message indicates that two copies of homebridge are running, the second copy is likely started by autostart scripts.

rsuard, 2019-05-24:
Thanks fou your answer @NorthernMan54 
Do you know the twist to kill them all ?

And why I have this at the end ?
 [2019-5-24 15:24:55] [PiTemp] Error: [Error: failed to read sensor]

rsuard, 2019-05-25:
I have found how to kill them all : sudo killall homebridge, then I followed this to reinstilize Homebridge : https://github.com/nfarina/homebridge/wiki/Basic-Troubleshooting

However, I ve no idea why my sensor is not read. I ll continue digging, if you have any idea ?

NorthernMan54, 2019-05-25:
Did you do this?
sudo adduser homebridge gpio

rsuard, 2019-05-29:
Hello,

I tried, but I run Homebridge with my "pi" user wich is in the GPIO group. I have no user called "homebridge", so when I put "sudo adduser homebridge gpio", it answers "user Homebridge doesn't exist"

What am I missing ?

NorthernMan54, 2019-05-29:
If your running as the user pi try

`sudo adduser pi gpio`

rsuard, 2019-05-31:
Hello NoternMan54,

Thanks, I already tried it and got the answer :
Pi is already in GPIO group

I m thinking about a hardware problem, is there a way to test it ?
Thanks

toshibochan, 2019-05-31:
I did new installation yesterday and not working to.

NorthernMan54, 2019-06-03:
@rsuard Under the covers I'm using this library 'node-dht-sensor' to read the temperature and the error is from it.

I was looking at their issues, and they had some very good troubleshooting in this issue

https://github.com/momenso/node-dht-sensor/issues/57

@toshibochan Please open a separate issue, and share a log file with an errors.

closed 2019-08-06 by NorthernMan54