NorthernMan54/homebridge-dht #33: ERROR LOADING PLUGIN homebridge-dht

tonykingnz, 2019-09-27:
pi@raspberrypi:~ $ homebridge
[2019-9-27 01:30:14] Loaded config.json with 2 accessories and 0 platforms.
[2019-9-27 01:30:14] ---
[2019-9-27 01:30:14] ====================
[2019-9-27 01:30:14] ERROR LOADING PLUGIN homebridge-dht:
[2019-9-27 01:30:14] Error: Cannot find module 'node-dht-sensor'
    at Function.Module._resolveFilename (module.js:543:15)
    at Function.Module._load (module.js:470:25)
    at Module.require (module.js:593:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge-dht/index.js:40:14)
    at Module._compile (module.js:649:30)
    at Object.Module._extensions..js (module.js:660:10)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:501:12)
    at Function.Module._load (module.js:493:3)
[2019-9-27 01:30:14] ====================
[2019-9-27 01:30:14] No plugins found. See the README for information on installing plugins.
[2019-9-27 01:30:14] Loading 0 platforms...
[2019-9-27 01:30:14] Loading 2 accessories...
[2019-9-27 01:30:14] Error: The requested accessory 'Dht' was not registered by any plugin.
    at API.accessory (/usr/local/lib/node_modules/homebridge/lib/api.js:64:13)
    at Server._loadAccessories (/usr/local/lib/node_modules/homebridge/lib/server.js:297:42)
    at Server.run (/usr/local/lib/node_modules/homebridge/lib/server.js:91:38)
    at module.exports (/usr/local/lib/node_modules/homebridge/lib/cli.js:59:10)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge/bin/homebridge:17:22)
    at Module._compile (module.js:649:30)
    at Object.Module._extensions..js (module.js:660:10)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:501:12)
    at Function.Module._load (module.js:493:3)
[2019-9-27 01:30:14] Got SIGTERM, shutting down Homebridge...

What can I do? Please! I need to do a work using it.

NorthernMan54, 2019-09-28:
> at Object. (/usr/local/lib/node_modules/homebridge-dht/index.js:40:14)
> at Object. (/usr/local/lib/node_modules/homebridge/bin/homebridge:17:22)

Looks like the plugin did not install incorrectly as the dependancy "node-dht-sensor” was not found

Can you try installing the plugin again, and watch for any errors during plugin installation

tonykingnz, 2019-10-01:
pi@raspberrypi:~ $ sudo npm install -g homebridge-dht
+ homebridge-dht@0.1.27
updated 1 package in 16.067s
pi@raspberrypi:~ $ homebridge
[2019-10-1 21:54:36] Loaded config.json with 2 accessories and 0 platforms.
[2019-10-1 21:54:36] ---
[2019-10-1 21:54:36] ====================
[2019-10-1 21:54:36] ERROR LOADING PLUGIN homebridge-dht:
[2019-10-1 21:54:36] Error: Cannot find module 'node-dht-sensor'
    at Function.Module._resolveFilename (module.js:543:15)
    at Function.Module._load (module.js:470:25)
    at Module.require (module.js:593:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge-dht/index.js:40:14)
    at Module._compile (module.js:649:30)
    at Object.Module._extensions..js (module.js:660:10)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:501:12)
    at Function.Module._load (module.js:493:3)
[2019-10-1 21:54:36] ====================
[2019-10-1 21:54:36] No plugins found. See the README for information on installing plugins.
[2019-10-1 21:54:36] Loading 0 platforms...
[2019-10-1 21:54:36] Loading 2 accessories...
[2019-10-1 21:54:36] Error: The requested accessory 'Dht' was not registered by any plugin.
    at API.accessory (/usr/local/lib/node_modules/homebridge/lib/api.js:64:13)
    at Server._loadAccessories (/usr/local/lib/node_modules/homebridge/lib/server.js:297:42)
    at Server.run (/usr/local/lib/node_modules/homebridge/lib/server.js:91:38)
    at module.exports (/usr/local/lib/node_modules/homebridge/lib/cli.js:59:10)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge/bin/homebridge:17:22)
    at Module._compile (module.js:649:30)
    at Object.Module._extensions..js (module.js:660:10)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:501:12)
    at Function.Module._load (module.js:493:3)
[2019-10-1 21:54:36] Got SIGTERM, shutting down Homebridge...

tonykingnz, 2019-10-01:
I reinstalled the dht plugin, but nothing happened.

tonykingnz, 2019-10-01:
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/pacote/node_modules/minimatch/node_modules/brace-expansion/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/pacote/node_modules/minimatch/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/pacote/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/pacote/node_modules/promise-retry/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/pacote/node_modules/protoduck/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/pacote/node_modules/tar-fs/node_modules/pump/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/pacote/node_modules/tar-fs/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/pacote/node_modules/tar-stream/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/read-installed/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/read-package-json/node_modules/json-parse-helpfulerror/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/read-package-json/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/read/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/readable-stream/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/request/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/request/node_modules/combined-stream/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/request/node_modules/form-data/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/request/node_modules/har-validator/node_modules/ajv/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/request/node_modules/har-validator/node_modules/ajv/node_modules/json-stable-stringify/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/request/node_modules/har-validator/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/request/node_modules/hawk/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/request/node_modules/http-signature/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/request/node_modules/http-signature/node_modules/jsprim/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/request/node_modules/http-signature/node_modules/sshpk/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/request/node_modules/mime-types/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/request/node_modules/tough-cookie/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/sorted-union-stream/node_modules/from2/node_modules/readable-stream/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/sorted-union-stream/node_modules/from2/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/sorted-union-stream/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/sorted-union-stream/node_modules/stream-iterate/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/strip-ansi/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/tar/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/unique-filename/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/update-notifier/node_modules/boxen/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/update-notifier/node_modules/boxen/node_modules/string-width/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/update-notifier/node_modules/boxen/node_modules/term-size/node_modules/execa/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/update-notifier/node_modules/boxen/node_modules/term-size/node_modules
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules/npm/node_modules/update-notifier/node_modules/boxen/node_modules/widest-line/node_modules/string-width/node_modules
[... cut ...]
npm ERR!   stack: 'Error: EACCES: permission denied, access \'/usr/local/lib/node_modules/npm/node_modules/abbrev\'',
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'access',
npm ERR!   path: '/usr/local/lib/node_modules/npm/node_modules/abbrev' }
npm ERR!
npm ERR! Please try running this command again as root/Administrator.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-10-01T21_49_33_672Z-debug.log
pi@raspberrypi:~ $

I tried to update npm but got this.

NorthernMan54, 2019-10-01:
Hum..

Can you remove homebridge-dht and try installing again, for some reason that dependency is not getting installed

tonykingnz, 2019-10-03:
How I do it?

NorthernMan54, 2019-10-03:
Find the directory it is installed in and remove it

Ie

sudo rm -rf /usr/local/lib/node_modules/homebridge-dht

ukjoci, 2019-10-07:
Hi There!

I have also a problem with running or loading dht plugin in Homebridge. Before it worked, but it broguht time to work.
But now, I get this error message and my DHT22 plugin doesn't work with Homebridge and iOS Home Kit.
The DHT22 sensor is good, because in Raspi terminal I can read out the values of temperature and humidity.
I tried uninstall, re-install, and so on, but it's not working. What couldf be wrong?
[image]

Here is the full error message from the terminal:

ERROR LOADING PLUGIN homebridge-dht:
[2019-10-6 22:11:04] Error: Cannot find module '/home/pi/homebridge-dht/node_modules/node-dht-sensor/build/Release/node_dht_sensor'. Please verify that the package.json has a valid "main" entry
at tryPackage (internal/modules/cjs/loader.js:288:19)
at Function.Module._findPath (internal/modules/cjs/loader.js:515:18)
at Function.Module._resolveFilename (internal/modules/cjs/loader.js:759:27)
at Function.Module._load (internal/modules/cjs/loader.js:677:27)
at Module.require (internal/modules/cjs/loader.js:830:19)
at require (internal/modules/cjs/helpers.js:68:18)
at Object. (/home/pi/homebridge-dht/index.js:40:14)
at Module._compile (internal/modules/cjs/loader.js:936:30)
at Object.Module._extensions..js (internal/modules/cjs/loader.js:947:10)
at Module.load (internal/modules/cjs/loader.js:790:32)
And here is a picture of it also.

Many-many thanks for your help!
dht_problem

tonykingnz, 2019-10-08:
I removed and tried install again. Happened the same thing. Then i used the command:

sudo npm install homebridge-dht --unsafe-perm

and got it:

pi@raspberrypi:/usr/local/lib/node_modules $ sudo npm install homebridge-dht --unsafe-perm

> node-dht-sensor@0.0.34 preinstall /usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor
> ./check-lib.sh

Library bcm2835 found.

> node-dht-sensor@0.0.34 install /usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor
> node-gyp configure

> node-dht-sensor@0.0.34 postinstall /usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor
> node-gyp build

make: Entering directory '/usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor/build'
  CXX(target) Release/obj.target/node_dht_sensor/node-dht-sensor.o
In file included from /root/.node-gyp/8.2.1/include/node/node.h:63,
                 from ../node-dht-sensor.cpp:1:
/root/.node-gyp/8.2.1/include/node/v8.h: In static member function ‘static void v8::V8::RemoveGCPrologueCallback(v8::GCCallback)’:
/root/.node-gyp/8.2.1/include/node/v8.h:9814:53: warning: cast between incompatible function types from ‘v8::GCCallback’ {aka ‘void (*)(v8::GCType, v8::GCCallbackFlags)’} to ‘v8::Isolate::GCCallback’ {aka ‘void (*)(v8::Isolate*, v8::GCType, v8::GCCallbackFlags)’} [-Wcast-function-type]
       reinterpret_cast<Isolate::GCCallback>(callback));
                                                     ^
/root/.node-gyp/8.2.1/include/node/v8.h: In static member function ‘static void v8::V8::RemoveGCEpilogueCallback(v8::GCCallback)’:
/root/.node-gyp/8.2.1/include/node/v8.h:9821:53: warning: cast between incompatible function types from ‘v8::GCCallback’ {aka ‘void (*)(v8::GCType, v8::GCCallbackFlags)’} to ‘v8::Isolate::GCCallback’ {aka ‘void (*)(v8::Isolate*, v8::GCType, v8::GCCallbackFlags)’} [-Wcast-function-type]
       reinterpret_cast<Isolate::GCCallback>(callback));
                                                     ^
In file included from ../node-dht-sensor.cpp:2:
../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
../../nan/nan.h:1875:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka  void (*)(uv_work_s*, int)’} [-Wcast-function-type]
     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
                                                              ^
In file included from ../node-dht-sensor.cpp:1:
../node-dht-sensor.cpp: At global scope:
/root/.node-gyp/8.2.1/include/node/node.h:460:43: warning: cast between incompatible function types from ‘void (*)(v8::Handle<v8::Object>)’ {aka ‘void (*)(v8::Local<v8::Object>)’} to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
       (node::addon_register_func) (regfunc),                          \
                                           ^
/root/.node-gyp/8.2.1/include/node/node.h:491:3: note: in expansion of macro ‘NODE_MODULE_X’
   NODE_MODULE_X(modname, regfunc, NULL, 0)
   ^~~~~~~~~~~~~
../node-dht-sensor.cpp:200:1: note: in expansion of macro ‘NODE_MODULE’
 NODE_MODULE(node_dht_sensor, Init);
 ^~~~~~~~~~~
In file included from /root/.node-gyp/8.2.1/include/node/node.h:63,
                 from ../node-dht-sensor.cpp:1:
/root/.node-gyp/8.2.1/include/node/v8.h: In instantiation of ‘void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackInfo<P>::Callback, v8::WeakCallbackType) [with P = node::ObjectWrap; T = v8::Object; typename v8::WeakCallbackInfo<P>::Callback = void (*)(const v8::WeakCallbackInfo<node::ObjectWrap>&)]’:
/root/.node-gyp/8.2.1/include/node/node_object_wrap.h:85:78:   required from here
/root/.node-gyp/8.2.1/include/node/v8.h:8793:16: warning: cast between incompatible function types from ‘v8::WeakCallbackInfo<node::ObjectWrap>::Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<node::ObjectWrap>&)’} to ‘Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<void>&)’} [-Wcast-function-type]
                reinterpret_cast<Callback>(callback), type);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/.node-gyp/8.2.1/include/node/v8.h: In instantiation of ‘void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackInfo<P>::Callback, v8::WeakCallbackType) [with P = Nan::ObjectWrap; T = v8::Object; typename v8::WeakCallbackInfo<P>::Callback = void (*)(const v8::WeakCallbackInfo<Nan::ObjectWrap>&)]’:
../../nan/nan_object_wrap.h:66:61:   required from here
/root/.node-gyp/8.2.1/include/node/v8.h:8793:16: warning: cast between incompatible function types from ‘v8::WeakCallbackInfo<Nan::ObjectWrap>::Callback’ {aka  void (*)(const v8::WeakCallbackInfo<Nan::ObjectWrap>&)’} to ‘Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<void>&)’} [-Wcast-function-type]
  CXX(target) Release/obj.target/node_dht_sensor/dht-sensor.o
  SOLINK_MODULE(target) Release/obj.target/node_dht_sensor.node
  COPY Release/node_dht_sensor.node
make: Leaving directory '/usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor/build'
+ homebridge-dht@0.1.27
added 70 packages in 105.662s

   ╭──────────────────────────────────────╮
   │                                      │
   │   Update available 5.3.0 → 6.12.0    │
   │      Run npm i -g npm to update      │
   │                                      │
   ╰──────────────────────────────────────╯

pi@raspberrypi:/usr/local/lib/node_modules $

NorthernMan54, 2019-10-08:
> /root/.node-gyp/8.2.1/include/node/v8.h: In instantiation of ‘void v8::PersistentBase::SetWeak(P*, typename v8::WeakCallbackInfo
> /root/.node-gyp/8.2.1/include/node/v8.h:8793:16: warning: cast between incompatible function types from ‘v8::WeakCallbackInfonode::ObjectWrap::Callback’ {aka ‘void ()(const v8::WeakCallbackInfonode::ObjectWrap&)’} to ‘Callback’ {aka ‘void ()(const v8::WeakCallbackInfo&)’} [-Wcast-function-type]
> reinterpret_cast(callback), type);
> /root/.node-gyp/8.2.1/include/node/v8.h: In instantiation of ‘void v8::PersistentBase::SetWeak(P, typename v8::WeakCallbackInfo
> /root/.node-gyp/8.2.1/include/node/v8.h:8793:16: warning: cast between incompatible function types from ‘v8::WeakCallbackInfoNan::ObjectWrap::Callback’ {aka void ()(const v8::WeakCallbackInfoNan::ObjectWrap&)’} to ‘Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo&)’} [-Wcast-function-type]

Did it work afterwards?

tonykingnz, 2019-10-08:
Got this

pi@raspberrypi:/usr/local/lib/node_modules $ homebridge
[2019-10-9 00:01:03] Loaded config.json with 2 accessories and 0 platforms.
[2019-10-9 00:01:03] ---
[2019-10-9 00:01:05] Loaded plugin: homebridge-dht
[2019-10-9 00:01:05] Registering accessory 'homebridge-dht.Dht'
[2019-10-9 00:01:05] ---
[2019-10-9 00:01:05] Loading 0 platforms...
[2019-10-9 00:01:05] Loading 2 accessories...
[2019-10-9 00:01:05] [cputemp] Initializing Dht accessory...
[2019-10-9 00:01:05] [cputemp] Adding Accessory
[2019-10-9 00:01:05] [cputemp] INIT: cputemp
[2019-10-9 00:01:05] [Temp/Humidity Sensor] Initializing Dht accessory...
[2019-10-9 00:01:05] [Temp/Humidity Sensor] Adding Accessory
[2019-10-9 00:01:05] [Temp/Humidity Sensor] INIT: Temp/Humidity Sensor
Setup Payload:
X-HM://0023ISYWY28LH
Scan this code with your HomeKit app on your iOS device to pair with Homebridge:

Or enter this code with your HomeKit app on your iOS device to pair with Homebridge:

    ┌────────────┐
    │ 031-45-154 │
    └────────────┘

[2019-10-9 00:01:05] Error: listen EADDRINUSE :::51826
    at Object.exports._errnoException (util.js:1024:11)
    at exports._exceptionWithHostPort (util.js:1047:20)
    at Server.setupListenHandle [as _listen2] (net.js:1319:14)
    at listenInCluster (net.js:1367:12)
    at Server.listen (net.js:1467:7)
    at EventedHTTPServer.listen (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/util/eventedhttp.js:60:19)
    at HAPServer.listen (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/HAPServer.js:158:20)
    at Bridge.Accessory.publish (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/lib/Accessory.js:616:16)
    at Server._publish (/usr/local/lib/node_modules/homebridge/lib/server.js:132:16)
    at Server.run (/usr/local/lib/node_modules/homebridge/lib/server.js:101:10)
[2019-10-9 00:01:05] Got SIGTERM, shutting down Homebridge...
[2019-10-9 00:01:07] [Temp/Humidity Sensor] Error: Error: failed to read sensor

pi@raspberrypi:/usr/local/lib/node_modules $

NorthernMan54, 2019-10-08:
Looks like it worked, you just have homebridge already running hence the already in use error

tonykingnz, 2019-10-08:
I restarted the raspberry:

pi@raspberrypi:~ $ homebridge
[2019-10-9 00:16:40] Loaded config.json with 2 accessories and 0 platforms.
[2019-10-9 00:16:40] ---
[2019-10-9 00:16:40] No plugins found. See the README for information on installing plugins.
[2019-10-9 00:16:40] Loading 0 platforms...
[2019-10-9 00:16:40] Loading 2 accessories...
[2019-10-9 00:16:40] Error: The requested accessory 'Dht' was not registered by any plugin.
    at API.accessory (/usr/local/lib/node_modules/homebridge/lib/api.js:64:13)
    at Server._loadAccessories (/usr/local/lib/node_modules/homebridge/lib/server.js:297:42)
    at Server.run (/usr/local/lib/node_modules/homebridge/lib/server.js:91:38)
    at module.exports (/usr/local/lib/node_modules/homebridge/lib/cli.js:59:10)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge/bin/homebridge:17:22)
    at Module._compile (module.js:569:30)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
[2019-10-9 00:16:40] Got SIGTERM, shutting down Homebridge...
pi@raspberrypi:~ $

NorthernMan54, 2019-10-09:
Looks like a problem with your config.json

tonykingnz, 2019-10-09:
pi@raspberrypi:~/.homebridge $ sudo nano /home/pi/.homebridge/config.json
  

GNU nano 3.2            /home/pi/.homebridge/config.json

{ "bridge": {
        "name": "Penny",
        "username": "CC:22:3D:E3:CD:33",
        "port": 51826,
        "pin": "031-45-154"
    },

"description": "HomeBridge",
"platforms": [],

"accessories": [
      { "accessory":          "Dht",
        "name":               "cputemp",
        "service":        "Temperature" },
      { "accessory":          "Dht",
        "name":               "Temp/Humidity Sensor",
        "service":            "dht22" }
    ]}

What is wrong?

NorthernMan54, 2019-10-09:
@tonykingnz I originally that the error was a typo in the config.json, but after reviewing your config.json it is okay.

Looking back, the error message is that no plugins were found this time, but I don’t know why.

 No plugins found. See the README for information on installing plugins.

I’m stumped to determine what is wrong with your setup.

tonykingnz, 2019-10-10:
Too bad you didn't find a solution. Thank you for taking the time to help me out. Have a good night. Where could I search to try to find the solution alone?

closed 2019-11-09 by NorthernMan54