NorthernMan54/homebridge-dht #37: Installation error on RPi3 with Buster

chimcen, 2019-11-09:
Hi,

thank you for your plugin - but i got an error during installation.
I‘ve followed your instruction, installed bcm lib before, but i got an erren when installing your plugin on my Pi with Raspian Buster.

Here are the errors:

`pi@Raspi:~ $ sudo npm install -g --unsafe-perm  homebridge-dht

> node-dht-sensor@0.0.34 preinstall /usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor
> ./check-lib.sh

Library bcm2835 found.

> node-dht-sensor@0.0.34 install /usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor
> node-gyp configure

> node-dht-sensor@0.0.34 postinstall /usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor
> node-gyp build

make: Verzeichnis „/usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor/build“ wird betreten
  CXX(target) Release/obj.target/node_dht_sensor/node-dht-sensor.o
In file included from ../../nan/nan_converters.h:67:0,
                 from ../../nan/nan.h:202,
                 from ../node-dht-sensor.cpp:2:
../../nan/nan_converters_43_inl.h: In static member function ‘static Nan::imp::ToFactoryBase<v8::Boolean>::return_t Nan::imp::ToFactory<v8::Boolean>::convert(v8::Local<v8::Value>)’:
../../nan/nan_converters_43_inl.h:18:69: warning: ‘v8::MaybeLocal<v8::Boolean> v8::Value::ToBoolean(v8::Local<v8::Context>) const’ is deprecated (declared at /root/.cache/node-gyp/12.13.0/include/node/v8.h:2567): ToBoolean can never throw. Use Local version. [-Wdeprecated-declarations]
       val->To ## TYPE(v8::Isolate::GetCurrent()->GetCurrentContext())          \
                                                                     ^
../../nan/nan_converters_43_inl.h:22:1: note: in expansion of macro ‘X’
 X(Boolean)
 ^
../../nan/nan_converters_43_inl.h: In static member function ‘static Nan::imp::ValueFactoryBase<bool>::return_t Nan::imp::ToFactory<bool>::convert(v8::Local<v8::Value>)’:
../../nan/nan_converters_43_inl.h:37:57: warning: ‘v8::Maybe<bool> v8::Value::BooleanValue(v8::Local<v8::Context>) const’ is deprecated (declared at /root/.cache/node-gyp/12.13.0/include/node/v8.h:2605): BooleanValue can never throw. Use Isolate version. [-Wdeprecated-declarations]
   return val->NAME ## Value(isolate->GetCurrentContext());                     \
                                                         ^
../../nan/nan_converters_43_inl.h:40:1: note: in expansion of macro ‘X’
 X(bool, Boolean)
 ^
In file included from ../../nan/nan_new.h:189:0,
                 from ../../nan/nan.h:203,
                 from ../node-dht-sensor.cpp:2:
../../nan/nan_implementation_12_inl.h: In static member function ‘static Nan::imp::FactoryBase<v8::Function>::return_t Nan::imp::Factory<v8::Function>::New(Nan::FunctionCallback, v8::Local<v8::Value>)’:
../../nan/nan_implementation_12_inl.h:105:32: error: no matching function for call to ‘v8::Function::New(v8::Isolate*&, void (&)(const v8::FunctionCallbackInfo<v8::Value>&), v8::Local<v8::Object>&)’
                           , obj));
                                ^
../../nan/nan_implementation_12_inl.h:105:32: note: candidate is:
In file included from /root/.cache/node-gyp/12.13.0/include/node/node.h:63:0,
                 from ../node-dht-sensor.cpp:1:
/root/.cache/node-gyp/12.13.0/include/node/v8.h:4170:31: note: static v8::MaybeLocal<v8::Function> v8::Function::New(v8::Local<v8::Context>, v8::FunctionCallback, v8::Local<v8::Value>, int, v8::ConstructorBehavior, v8::SideEffectType)
   static MaybeLocal<Function> New(
                               ^
/root/.cache/node-gyp/12.13.0/include/node/v8.h:4170:31: note:   no known conversion for argument 1 from ‘v8::Isolate*’ to ‘v8::Local<v8::Context>’
In file included from ../../nan/nan_new.h:189:0,
                 from ../../nan/nan.h:203,
                 from ../node-dht-sensor.cpp:2:
../../nan/nan_implementation_12_inl.h: In static member function ‘static Nan::imp::FactoryBase<v8::StringObject>::return_t Nan::imp::Factory<v8::StringObject>::New(v8::Local<v8::String>)’:
../../nan/nan_implementation_12_inl.h:337:37: error: no matching function for call to ‘v8::StringObject::New(v8::Local<v8::String>&)’
   return v8::StringObject::New(value).As<v8::StringObject>();
                                     ^
../../nan/nan_implementation_12_inl.h:337:37: note: candidate is:
In file included from /root/.cache/node-gyp/12.13.0/include/node/node.h:63:0,
                 from ../node-dht-sensor.cpp:1:
/root/.cache/node-gyp/12.13.0/include/node/v8.h:5426:23: note: static v8::Local<v8::Value> v8::StringObject::New(v8::Isolate*, v8::Local<v8::String>)
   static Local<Value> New(Isolate* isolate, Local<String> value);
                       ^
/root/.cache/node-gyp/12.13.0/include/node/v8.h:5426:23: note:   candidate expects 2 arguments, 1 provided
In file included from ../../nan/nan_new.h:189:0,
                 from ../../nan/nan.h:203,
                 from ../node-dht-sensor.cpp:2:
[... 215 lines cut ...]
                  ^
make: *** [node_dht_sensor.target.mk:110: Release/obj.target/node_dht_sensor/node-dht-sensor.o] Fehler 1
make: Verzeichnis „/usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor/build“ wird verlassen
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:210:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Linux 4.19.75-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build"
gyp ERR! cwd /usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor
gyp ERR! node -v v12.13.0
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-dht-sensor@0.0.34 postinstall: `node-gyp build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-dht-sensor@0.0.34 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-11-09T10_15_15_570Z-debug.log
`

NorthernMan54, 2019-11-09:
The issue is with the installation of this module, https://github.com/momenso/node-dht-sensor and I think I have it grabbing an older version.

Let me look into it.

NorthernMan54, 2019-11-11:
I updated the dependant libraries to the latest version, please re-install the plugin and try again.

chimcen, 2019-11-13:
Awesome - it works like a charn.

Some questions:
1) is it possible to combine temp and hunidity to one"sensor" - just to display the dht22 the same way as the BME280 from your mcuiot project in Apple Home or Eve App?

2) The refresh time is in minutes or seconds entered in config.json?

3) is it possible to store all meassurements from the dht22 on the raspi like its done at the bme's from mcuiot project to display history graphs in eve app?

Thank you :)

NorthernMan54, 2019-11-14:
Some questions:

1. is it possible to combine temp and humidity to one"sensor" - just to display the dht22 the same way as the BME280 from your mcuiot project in Apple Home or Eve App?

_Are you referring to the iOS 13 behaviour?_  
[image]

_Basement is a DHT22, and Cold Cellar is a MCUIOT_

2. The refresh time is in minutes or seconds entered in config.json?
 
_It is in seconds, I fixed the README me to include this_

3. Is it possible to add an +/- offset to the meassurements of the dht22? eventually directly in config.json?

_That is not a current feature, if you are looking for accuracy you may want to use the BME280 instead.  The DHT22 are horribly inaccurate.  But if you really want it, I would gladly accept a pull request with this a new optional feature._

NorthernMan54, 2019-11-14:
After looking at the iOS 13 behaviour, I made it match the MCUIOT behaviour

chimcen, 2019-11-14:
It works - thank you :)

To point 3) - ...I’ve actually no idea how to do this...

closed 2019-11-22 by NorthernMan54