NorthernMan54/homebridge-dht #38: Not showing humidity and temperature as two sensors on home screen (DHT22)

knopfy, 2019-11-21:
Yesterday I set up a new raspberry 3, installed homebridge and the homebridge-dht plugin. Everything works well so far. But on the home screen only temperature shows up. I can see humidity if I click on edit and then on the sensor icon. How can I get both temperature AND humidity readable from the home screen icon?

toshibochan, 2019-11-21:
In the home app you can separate in the accessory setting.

acoolax, 2019-11-21:
> In the home app you can separate in the accessory setting.

How? I didn't find it :-(

NorthernMan54, 2019-11-22:
With the changes in accessory grouping behaviour with iOS 13, I had to make some changes.  And what your seeing is the current behaviour

The new behaviour

[image]

The Icon on the right is what happened with iOS 13, and the left icons are the current behaviour.

[image]

closed 2019-11-22 by NorthernMan54

Penpal1278, 2019-11-27:
Hi, based on the previous comments, I understand that there were some changes made to accommodate changes in iOS 13, but is there still a way to separately display temperature and humidity in the Home app as two different sensors?  (Currently using this plugin with a DHT11; I only see a single sensor in my Home app, which displays current temp on its icon)

NorthernMan54, 2019-11-28:
Nope unless Apple changes there grouping