NorthernMan54/homebridge-dht #39: Cant get it to work with DHT11

masud321, 2019-11-22:
Doesn't seem to be working with DHT11. 
The output remains (stuck) at 0.

NorthernMan54, 2019-11-24:
Can you validate the wiring, by trying the sample here

https://github.com/momenso/node-dht-sensor#first-example

This is the library I’m using to connect to the device

I suspect I may have a code problem, as I don’t have a dht11 to test with

masud321, 2019-11-25:
The wiring is correct, tried different variations but nothing is working. I noticed "Service" must be "dht22" or I can just not include the line, otherwise "dht11" for "service" won't run. 
I think dht11 is not part of your code maybe?

NorthernMan54, 2019-11-26:
Update and try again.  I added a line for the dht11

NorthernMan54, 2019-11-26:
Service should be dht11 for the Dht11

masud321, 2019-11-26:
You are the MAN!
"gpio" should be specified and now its working! 
Thanks buddy!

NorthernMan54, 2019-11-26:
Glad to hear

closed 2019-11-26 by NorthernMan54