NorthernMan54/homebridge-dht #43: cputemp is not working

neo3, 2019-12-18:
config.json
`{
            "accessory": "Dht",
            "name": "cputemp",
            "service": "Temperature"
        }`

Log
`[2019-12-18 3:03:44 PM] [Sensor] DHT Status: 0, Temperature: 27.2°C, Humidity: 51.6%
[2019-12-18 3:03:44 PM] [cputemp] Got Temperature of NaN`

Test in the Terminal
`pi@afccasa:~ $ /usr/local/bin/cputemp
CPU temperature: 58.534 °C`

NorthernMan54, 2019-12-19:
The output from cputemp isn't correct, it should look like this

```
cputemp
42 C
```

neo3, 2019-12-19:
config.json
> {
>             "accessory": "Dht",
>             "name": "Raspberry",
>             "service": "Temperature",
>             "cputemp": "/usr/local/bin/temperatura.sh"
>         }

Test in the Terminal
> pi@afccasa:/usr/local/bin $ /usr/local/bin/temperatura.sh
> 60 C

The same error still happens.

NorthernMan54, 2019-12-19:
Not sure what's happening on your machine....

test.js
```
var exec = require('child_process').execFile;
    exec("cputemp", function(error, responseBody, stderr) {
      if (error !== null) {
        console.log('cputemp function failed: ' + error);
      } else {
        var binaryState = parseFloat(responseBody);
        console.log("Got Temperature of %s", binaryState);

      }
    }.bind(this));
```

```
node test.js 
Got Temperature of 45
```

neo3, 2019-12-19:
In the index.js file cputemp was not quoted and after some testing I saw that the function was not getting my custom script. So I modified it for this:

> exec("/home/pi/cputemp", function(error, responseBody, stderr)

NorthernMan54, 2019-12-21:
With this workaround, is it working for you?

neo3, 2019-12-21:
Yes, thanks!

closed 2019-12-21 by neo3