NorthernMan54/homebridge-dht #44: DHT22 does not work, show as 0C and 0%.

nuttdam, 2020-05-27:
I have install BCM2835 (downloaded at /home/pi) and homebridge-dht both from config-ui-x and using command line with `--unsafe-perm` and this is my config.json
`{
            "accessory": "Dht",
            "name": "Room",
            "service": "dht22"
        }`
but everything shows as 0 celcius and 0% humidity.

I am connecting DHT22 at GPIO4 (pin7) and I think I have follow through every steps, any suggestion?

Thanks

NorthernMan54, 2020-05-27:
Do the homebridge logs show any errors?

nuttdam, 2020-05-28:
> Do the homebridge logs show any errors?
> […](#)

`[5/28/2020, 10:52:54 AM] [Room] Initializing Dht accessory...
[5/28/2020, 10:52:54 AM] [Room] Adding Accessory
[5/28/2020, 10:52:54 AM] [Room] INIT: Room`

This is the only log I got.
By the way I try to manage this to work by adding refresh interval setting to json with value of 1.
`{
            "accessory": "Dht",
            "name": "Room",
            "gpio": "4",
            "service": "dht22",
            "refresh": "1"
        }`
Now the log shows
`[5/28/2020, 10:55:58 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:55:58 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:55:58 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:55:58 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:55:58 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:55:58 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:55:58 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:55:58 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:55:58 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:55:58 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:55:58 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:55:58 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:55:59 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:56:00 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%
[5/28/2020, 10:56:01 AM] [Room] DHT Status: 0, Temperature: 28.9°C, Humidity: 77.9%`
But I think this is too much for refreshing the sensor and when it works like this, in my Home app, the temp and humidity shows correctly but in Config UI X Accessories tab, all accessories is missing.

nuttdam, 2020-05-28:
It just seems to be able to capture temp and humidity but still not working correctly.

> [5/28/2020, 8:25:08 PM] [Room] DHT Status: 0, Temperature: 26.9°C, Humidity: 70.2%
[5/28/2020, 8:27:51 PM] [Room] DHT Status: 0, Temperature: 25.6°C, Humidity: 65.9%

Since Homebridge started at 11AM (server time), there are only 2 logs above. My current config is
`"refresh": "15"`

NorthernMan54, 2020-05-28:
This is really odd as the code is pretty simple and straight forward.  Is there anything else running on the RPI that could be blocking node js from reading the temperature.

nuttdam, 2020-05-29:
> This is really odd as the code is pretty simple and straight forward. Is there anything else running on the RPI that could be blocking node js from reading the temperature.

I have homebridge, homebridge_wol, homebridge_raspberrypi_temperature, transmission daemon and samba running on my pi.

NorthernMan54, 2020-05-29:
I’m stuck on where to go here

The code is pretty specific around the timing of reading a temperature.  And if your setup isn’t having issue’s I don’t know where to start

nuttdam, 2020-05-30:
I will try reinstall Raspbian and install only Homebridge. Will update to you later. Thanks,

agbb, 2020-06-06:
I am also seeing this issue. I am able to get a reading from the DHT using other libraries.

NorthernMan54, 2020-06-07:
Please check you config and which pin the various libraries expect to have the sensor connected to.

nuttdam, 2020-06-11:
I have tried new clean install with only samba running and mounting one external hdd. My config is
```
"accessories": [
        {
            "accessory": "Dht",
            "name": "Room"
        }
    ],
```
The result is still 0C and 0% same as before.

nuttdam, 2020-06-11:
The cputemp also not working.
```
"accessories": [
        {
            "accessory": "Dht",
            "name": "CPU Temperature",
            "service": "Temperature",
            "cputemp": "/usr/local/bin/cputemp"
        },
        {
            "accessory": "Dht",
            "name": "Room",
            "gpio": "4"
        }
    ]
```
This is what I get from cputemp.
```
pi@raspberrypi:~ $ /usr/local/bin/cputemp
47 C
pi@raspberrypi:~ $ cputemp
47 C
```

NorthernMan54, 2020-06-11:
Any logs from homebridge showing any errors?

nuttdam, 2020-06-12:
> Any logs from homebridge showing any errors?
> […](#)

No errors was shown, only this
```
[6/11/2020, 14:19:36] [Room] Initializing Dht accessory...
[6/11/2020, 14:19:36] [Room] Adding Accessory
[6/11/2020, 14:19:36] [Room] INIT: Room
```

NorthernMan54, 2020-06-12:
Without logs or other information my ability to troubleshoot is limited and have no idea why this isn’t working.

closed 2020-06-27 by NorthernMan54