NorthernMan54/homebridge-dht #45: Cannot read the sensor

blackLotus1998, 2020-08-08:
console showing 

Error: [Error: failed to read sensor]

NorthernMan54, 2020-08-11:
Not sure how to debug this, is it wired correctly?

aidzer0, 2020-08-14:
I’m having the same issue. Got 2 sensors, one sensor keeps going into “cannot read sensor” state, I unplug the + to the sensor and plug it back in, it reads, but a couple of hours later same issue occurs.

aidzer0, 2020-08-20:
The logs show the below, after unplugging and plugging the positive to the bedroom sensor, it re-establishes communication.

```[8/20/2020, 08:50:31] [dht22 - Bedroom] Error: [Error: failed to read sensor]
[8/20/2020, 08:50:32] [dht22 - Lounge] DHT Status: 0, Temperature: 24.4°C, Humidity: 57.6%
[8/20/2020, 08:51:32] [dht22 - Bedroom] Error: [Error: failed to read sensor]
[8/20/2020, 08:51:33] [dht22 - Lounge] DHT Status: 0, Temperature: 24.4°C, Humidity: 57.5%
[8/20/2020, 08:52:31] [dht22 - Bedroom] Error: [Error: failed to read sensor]
[8/20/2020, 08:52:32] [dht22 - Lounge] DHT Status: 0, Temperature: 24.4°C, Humidity: 57.7%
[8/20/2020, 08:53:29] [dht22 - Bedroom] DHT Status: 0, Temperature: 23°C, Humidity: 60%```

JohnWickham, 2020-09-19:
Having the same problem; been working perfectly for several months, and after a reboot it continuously fails to read the sensor. I've double-checked wiring and re-installed the plugin. Here’s the relevant portion of my config.json:

`       {
            "accessory": "Dht",
            "service": "dht22",
            "name": "Kitchen",
            "gpio": 4
         }
`

NorthernMan54, 2020-09-20:
@JohnWickham I'm just calling this library for the temperature http://www.airspayce.com/mikem/bcm2835/ do they have any comments on troubleshooting?

closed 2020-11-21 by NorthernMan54