NorthernMan54/homebridge-dht #46: Error on install

mikekenyon99, 2020-08-18:
Hi - I have  working Homebridge Pi Zero but when I install this package I get the following error (see screenshot).  I have installed other plugin's and they worked fine.   I'm sure it's my mistake but I can't work it out. 

Thanks

Mike 

[image]

NorthernMan54, 2020-08-18:
Try this instead

```
sudo su -
sudo npm install -g homebridge-dht
```

This is a NPM/NODE-GYP issue and not really a plugin issue

mikekenyon99, 2020-08-18:
Hi apologies if it's not the plugin's issue - I'm still getting the same issue though.  I will look it up and if I find a solution will post back here to help other people.

NorthernMan54, 2020-08-18:
There is this on google

https://github.com/aws-amplify/amplify-cli/issues/2023

mikekenyon99, 2020-08-19:
Worked a treat that.  Thanks.

closed 2020-08-19 by mikekenyon99