NorthernMan54/homebridge-dht #47: [dht22] Error: [Error: failed to read sensor]

lwitzani, 2021-01-16:
Hi, hope you are still around :) thx for the plugin.

I can't get it to work :(. 

I installed the homebridge plugin via homebridge-config-ui-x. My config is:

```
       {  
            "accessory": "Dht",  
            "name": "dht22",  
            "name_temperature": "Temperature",  
            "name_humidity": "Humidity",  
            "service": "dht22"   
        },
```

I installed the BCM2835 library etc and i can run dht22 in my terminal as normal pi user and also as homebridge user. It returns something like 
```
pi@raspberrypi:~ $ dht22    
0 21.5 C 49.9 %
```

When i boot my homebridge, the init stuff is 
```
[1/16/2021, 5:19:05 PM] [dht22] Initializing Dht accessory...
[1/16/2021, 5:19:05 PM] [dht22] Adding Accessory
[1/16/2021, 5:19:05 PM] [dht22] INIT: dht22
```

But later when it actually runs it there is the error
```
[1/16/2021, 5:20:35 PM] [dht22] Error: [Error: failed to read sensor]
```

Any idea what this could be?

NorthernMan54, 2021-01-16:
Without coming onsite and taking a look, my guess is account permissions.  And the account you are using on the command line is different than the one running homebridge.

lwitzani, 2021-01-16:
I thought the same, that is why i tried it with the homebridge user as following:
```
pi@raspberrypi:~ $ sudo su homebridge
homebridge@raspberrypi:/home/pi $ dht22
0 21.8 C 46.0 %
homebridge@raspberrypi:/home/pi $ sudo dht22
0 21.9 C 46.0 %

```

I made sure that homebridge-config-ui-x has all the rights it can get...
The homebridge user now even has passwordless sudo. I used 
`sudo visudo`
and added the line 
`homebridge    ALL=(ALL) NOPASSWD: ALL`

But still same problem...nothing changed

lwitzani, 2021-01-16:
is the dht22 command actually used? I can't find its usage in the index.js. There you use sensor.read( from the node-dht-sensor lib. I took a closer look to that and noticed if i run the code 
```
var sensor = require("../lib").promises;

// You can use `initialize` and `setMaxTries` just like before
sensor.setMaxRetries(10);
sensor.initialize(22, 4);

// You can still use the synchronous version of `read`:
// var readout = sensor.readSync(22, 4);

sensor.read(22, 4).then(
  function(res) {
    console.log(
      `temp: ${res.temperature.toFixed(1)}°C, ` +
        `humidity: ${res.humidity.toFixed(1)}%`
    );
  },
  function(err) {
    console.error("Failed to read sensor data:", err);
  }
);
```

copied from https://github.com/momenso/node-dht-sensor then i also get as result 
```
pi@raspberrypi:/usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor/lib $ sudo node fake.js 
Failed to read sensor data: [Error: failed to read sensor]
```
and the part "Error: failed to read sensor" is the exact message i get in homebridge so maybe there is a problem with this lib?

lwitzani, 2021-01-16:
I've rewritten (quick and dirty) some of your code so the dht22 executable is actually used...now it works
The part i changed is:
```
DhtAccessory.prototype = {

    getDHTTemperature: function (callback) {
        exec('dht22',
            function (error, stdout, stderr) {
                if (!error) {
                    var splitInfo = stdout.split(' ');
                    temperature = splitInfo[1];
                    humidity = splitInfo[3];
                    this.log("DHT Status: %s, Temperature: %s°C, Humidity: %s%", 0, roundInt(temperature), roundInt(humidity));

                    this.humidityService
                        .getCharacteristic(Characteristic.CurrentRelativeHumidity).updateValue(roundInt(humidity));
                    callback(error, roundInt(temperature));
                } else {
                    this.log.error("Error:", error);
                    callback(error);
                }
            }.bind(this));
    },
```

Will check tomorrow if i can further improve it for myself. 
Thanks anyway! 
Cheers

NorthernMan54, 2021-01-16:
If you go back thru the version history about 18-24 months ago I switched from this approach ( of calling an exec ) to the library as a lot of people had issues with getting the binary working.  The included binary is from the old version of the code.

franicrespo90, 2021-02-01:
hi i have the same problem, i was using a dht11 and it worked perfect, now i use a dht22 (it is more accurate) and i get: “[dht22] Error: [Error: failed to read sensor]” have you found any solution to this problem?  (I apologize I am a very beginner)

lwitzani, 2021-02-01:
As i mentioned, i used the code i wrote myself (which is similar to his actual code from some months ago). Works for me. Sry

franicrespo90, 2021-02-01:
perfect, I'll try your code, thank you very much

franicrespo90, 2021-02-02:
Good evening @lwitzani , I have tried to paste your code in index.js replacing the original code with yours, but when restarting the raspberry and running homebridge again I get this error:

`[2021-2-1 21:09:41] [dht22] Error: Error: spawn dht22 ENOENT
    at Process.ChildProcess._handle.onexit (internal/child_process.js:267:19)
    at onErrorNT (internal/child_process.js:469:16)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  errno: 'ENOENT',
  code: 'ENOENT',
  syscall: 'spawn dht22',
  path: 'dht22',
  spawnargs: [],
  cmd: 'dht22'
}`

maybe if we can solve it together we could update index.js in the plugin, where can I be failing? I apologize for my little knowledge

oscar-c, 2022-03-31:
The root cause of this problem is mentioned at momenso/node-dht-sensor#116 .
To fix this, you'll first need to go to `/usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor/src`, apply the change [Here](https://github.com/cainpsycode/node-dht-sensor/commit/50811e3ef593ed9b91a1fa2d17a1b4773bde549b),
Then go to `/usr/local/lib/node_modules/homebridge-dht/node_modules/node-dht-sensor`, run `node-gyp configure` and `node-gyp build`.
Then restart the home-assistant, you should have readings now.