NorthernMan54/homebridge-dht #51: Warning: characteristic 'Current Temperature':

truthin24, 2021-03-25:
Hello, thank you for all the effort. I have an issue I cannot seem to sort. I am running on a Raspberry Pi Zero.

The sensor is functional via dht22 command, but the errors in homebridge are still more common than readings. Readings are sucessful every 20-100 trys. 

Homebridge version v1.3.4 

Any thoughts would be much appreciated.

here is a copy of the diagnoistic log:

[3/25/2021, 8:54:26 PM] Homebridge v1.3.4 (Homebridge 1B9B) is running on port 51843.
[3/25/2021, 8:54:26 PM] [Room] read data from Room : {"firstEntry":0,"lastEntry":268,"usedMemory":268,"refTime":638313503,"initialTime":1616620703,"history":["noValue",{"time":1616620703,"setRefTime":1},{"temp":23.7,"humidity":43.1,"time":1616620703}

[3/25/2021, 8:54:26 PM] [Room] History Loaded from Persistant Storage
[3/25/2021, 8:54:27 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 8:54:29 PM] [homebridge-dht] This plugin generated a warning from the characteristic 'Current Temperature': characteristic value expected valid finite number and received "undefined" (undefined). See https://git.io/JtMGR for more info.
[3/25/2021, 8:54:29 PM] [homebridge-dht] Error: 
    at CurrentTemperature.Characteristic.characteristicWarning (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2038:105)
    at CurrentTemperature.Characteristic.validateUserInput (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1931:16)
    at CurrentTemperature.Characteristic.setValue (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:1298:20)
    at TemperatureSensor.Service.setCharacteristic (/usr/local/lib/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Service.ts:459:35)
    at DhtAccessory.<anonymous> (/usr/local/lib/node_modules/homebridge-dht/index.js:193:14)
    at DhtAccessory.<anonymous> (/usr/local/lib/node_modules/homebridge-dht/index.js:103:9)
[3/25/2021, 8:54:57 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 8:55:27 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 8:55:57 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 8:56:27 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 8:56:57 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 8:57:27 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 8:57:57 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 8:58:27 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 8:58:57 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 8:59:27 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 8:59:57 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 9:00:27 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 9:00:57 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 9:01:27 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 9:01:57 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 9:02:27 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 9:02:57 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 9:03:27 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 9:03:56 PM] [Room] DHT Status: 0, Temperature: 24.4°C, Humidity: 33.5%
[3/25/2021, 9:03:56 PM] [Room] **Fakegato-timer: addData  Room { time: 1616702636, temp: 24.4, humidity: 33.5 }  immediate:  false
[3/25/2021, 9:03:56 PM] [Room] **Start Global Fakegato-Timer - 5min**
[3/25/2021, 9:04:27 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 9:04:57 PM] [Room] Error: [Error: failed to read sensor]
[3/25/2021, 9:05:27 PM] [Room] Error: [Error: failed to read sensor]

truthin24, 2021-04-07:
here is the fix for the characterisitc issue. However, the sensor still reads only sporatiacally. 

        this.humidityService
          .getCharacteristic(Characteristic.CurrentRelativeHumidity).updateValue(roundInt(humidity));
        callback(err, roundInt(temperature));
      } else {
        this.log.error("Error:", err);
        **callback(err, 0); //added 0 to complete the callback**

NorthernMan54, 2021-04-07:
Are you sure your not having a hardware issue with your setup ?  ie device or wiring

In my similar setup ( I have 4 of these running ) I don't have the same issue