NorthernMan54/homebridge-dht #52: [dht22] 32-bit executable, won't run on aarch64

drwsctt, 2021-04-16:
I recently moved to the arm64 beta of Raspberry Pi OS, and when I tried to setup my Homebridge again I found that the dht22 executable is 32 bit, dynamically linked, and won't run on the 64-bit OS...

/usr/local/bin/dht22: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=7368cc48b57ad45ff60eb7cef27bd0e9719a5ae3, not stripped

It looks like you're just providing the binary (no source). Is there any chance you could recompile for arm64 or provide source so that I might build it myself?

Thanks!

NorthernMan54, 2021-04-16:
Due to the challenges with supporting the binary ( ie your issue ), I switched to a library based approach a few years ago.  The latest version uses a library and not longer uses a binary.

For the older version with the binary and build, if you look at the older version of the github repo, details are still there

https://github.com/NorthernMan54/homebridge-dht/tree/51c527c272e44dddac87bf15bc6470e16620a792

drwsctt, 2021-04-16:
Oh, sorry... I was just trying to move my old install over and failed to notice you had updated that. I'm all set now.

Thanks!

closed 2021-04-16 by drwsctt