NorthernMan54/homebridge-dht #53: No plug-in found for accessory “dht22” in config.json

Spudaaa, 2022-04-28:
Hi, not sure if the repos is still active but I’ve not long been into homebridge. Having a lot of issues. Managed to get this far from your instructable page.

keep getting these massages in the logs, any idea?

:
[28/04/2022, 21:33:07] Homebridge v1.4.0 (HAP v0.10.0) (Homebridge 855C) is running on port 51125.
[28/04/2022, 21:39:49] [Homebridge UI] Changes to config.json saved.
[28/04/2022, 21:40:06] [Homebridge UI] Homebridge restart request received
[28/04/2022, 21:40:06] [Homebridge UI] UI / Bridge settings have not changed; only restarting Homebridge process
[28/04/2022, 21:40:06] [Homebridge UI] Sending SIGTERM to Homebridge
[28/04/2022, 21:40:06] Got SIGTERM, shutting down Homebridge...
[28/04/2022, 21:40:11] [HB Supervisor] Homebridge Process Ended. Code: 143, Signal: null
[28/04/2022, 21:40:16] [HB Supervisor] Restarting Homebridge...
[28/04/2022, 21:40:16] [HB Supervisor] Starting Homebridge with extra flags: -I
[28/04/2022, 21:40:16] [HB Supervisor] Started Homebridge v1.4.0 with PID: 2130
[28/04/2022, 21:40:18] Loaded config.json with 2 accessories and 1 platforms.
[28/04/2022, 21:40:18] Loaded 0 cached accessories from cachedAccessories.
[28/04/2022, 21:40:18] ---
[28/04/2022, 21:40:20] Plugin /usr/local/lib/node_modules/homebridge-config-ui package.json does not contain the keyword 'homebridge-plugin'.
[28/04/2022, 21:40:21] Loaded plugin: homebridge-config-ui-x@4.43.1
[28/04/2022, 21:40:21] Registering platform 'homebridge-config-ui-x.config'
[28/04/2022, 21:40:21] ---
[28/04/2022, 21:40:21] Loading 1 platforms...
[28/04/2022, 21:40:21] [Config] Initializing config platform...
[28/04/2022, 21:40:21] [Config] Running in Service Mode
[28/04/2022, 21:40:21] Loading 2 accessories...
[28/04/2022, 21:40:21] No plugin was found for the accessory "Dht" in your config.json. Please make sure the corresponding plugin is installed correctly.
[28/04/2022, 21:40:21] No plugin was found for the accessory "Dht" in your config.json. Please make sure the corresponding plugin is installed correctly.
Setup Payload:
X-HM://0024VDJCJGNWM
Enter this code with your HomeKit app on your iOS device to pair with Homebridge:

Spudaaa, 2022-04-28:
I have reinstalled the plugin multiple times, checked the config.JSON seems to match your examples.
Desperate to get this first raspberry pi project working.

NorthernMan54, 2022-04-28:
That instructable is out of date and doesn't match the current configuration

Updated instructions are here - https://github.com/NorthernMan54/homebridge-dht/blob/master/Build.md

But with that said, this error means that the plugin did not install `No plugin was found for the accessory "Dht" in your config.json.`

Did you get any errors during the install ?

Spudaaa, 2022-04-29:
No errors installed perfectly.

Spudaaa, 2022-04-29:
I checked the GitHub instructions too still no joy. I have also reinstalled the plugin multiple times

Spudaaa, 2022-04-29:
i have completely flashed the imageback to the pi and started again, now when installing bm2835

pi@homebridge:/var/lib/homebridge/bcm2835-1.71 $ sudo make check
Making check in src
make[1]: Entering directory '/var/lib/homebridge/bcm2835-1.71/src'
make  test
make[2]: Entering directory '/var/lib/homebridge/bcm2835-1.71/src'
make[2]: 'test' is up to date.
make[2]: Leaving directory '/var/lib/homebridge/bcm2835-1.71/src'
make  check-TESTS
make[2]: Entering directory '/var/lib/homebridge/bcm2835-1.71/src'
make[3]: Entering directory '/var/lib/homebridge/bcm2835-1.71/src'
FAIL: test
============================================================================
Testsuite summary for bcm2835 1.71
============================================================================
# TOTAL: 1
# PASS:  0
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See src/test-suite.log
Please report to mikem@airspayce.com
============================================================================
make[3]: *** [Makefile:669: test-suite.log] Error 1
make[3]: Leaving directory '/var/lib/homebridge/bcm2835-1.71/src'
make[2]: *** [Makefile:777: check-TESTS] Error 2
make[2]: Leaving directory '/var/lib/homebridge/bcm2835-1.71/src'
make[1]: *** [Makefile:851: check-am] Error 2
make[1]: Leaving directory '/var/lib/homebridge/bcm2835-1.71/src'
make: *** [Makefile:365: check-recursive] Error 1

NorthernMan54, 2022-04-29:
That message is saying that the previous step `make` had an issue, what does the output of make say ?

Spudaaa, 2022-04-29:
start to failure is as follows

i@homebridge:/var/lib/homebridge $ tar zxvf bcm2835-1.71.tar.gz
bcm2835-1.71/
bcm2835-1.71/configure.ac
bcm2835-1.71/COPYING
bcm2835-1.71/examples/
bcm2835-1.71/examples/spi/
bcm2835-1.71/examples/spi/spi.c
bcm2835-1.71/examples/input/
bcm2835-1.71/examples/input/input.c
bcm2835-1.71/examples/spiram/
bcm2835-1.71/examples/spiram/spiram.h
bcm2835-1.71/examples/spiram/spiram_test.c
bcm2835-1.71/examples/spiram/spiram.c
bcm2835-1.71/examples/event/
bcm2835-1.71/examples/event/event.c
bcm2835-1.71/examples/pwm/
bcm2835-1.71/examples/pwm/pwm.c
bcm2835-1.71/examples/gpio/
bcm2835-1.71/examples/gpio/gpio.c
bcm2835-1.71/examples/blink/
bcm2835-1.71/examples/blink/blink.c
bcm2835-1.71/examples/spin/
bcm2835-1.71/examples/spin/spin.c
bcm2835-1.71/examples/i2c/
bcm2835-1.71/examples/i2c/i2c.c
bcm2835-1.71/config.guess
bcm2835-1.71/NEWS
bcm2835-1.71/depcomp
bcm2835-1.71/configure
bcm2835-1.71/aclocal.m4
bcm2835-1.71/compile
bcm2835-1.71/INSTALL
bcm2835-1.71/Makefile.in
bcm2835-1.71/missing
bcm2835-1.71/config.sub
bcm2835-1.71/Makefile.am
bcm2835-1.71/ChangeLog
bcm2835-1.71/README
bcm2835-1.71/install-sh
bcm2835-1.71/config.h.in
bcm2835-1.71/src/
bcm2835-1.71/src/test.c
bcm2835-1.71/src/bcm2835.h
bcm2835-1.71/src/bcm2835.c
bcm2835-1.71/src/Makefile.in
bcm2835-1.71/src/Makefile.am
bcm2835-1.71/ltmain.sh
bcm2835-1.71/AUTHORS
bcm2835-1.71/doc/
bcm2835-1.71/doc/Doxyfile.in
bcm2835-1.71/doc/Makefile.in
bcm2835-1.71/doc/Makefile.am
bcm2835-1.71/test-driver
pi@homebridge:/var/lib/homebridge $ cd bcm2835-1.71
pi@homebridge:/var/lib/homebridge/bcm2835-1.71 $ ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for clock_gettime in -lrt... yes
checking for doxygen... no
configure: WARNING: Doxygen not found - continuing without Doxygen support
checking for ranlib... ranlib
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating doc/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
pi@homebridge:/var/lib/homebridge/bcm2835-1.71 $ make
make  all-recursive
make[1]: Entering directory '/var/lib/homebridge/bcm2835-1.71'
Making all in src
make[2]: Entering directory '/var/lib/homebridge/bcm2835-1.71/src'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/var/lib/homebridge/bcm2835-1.71/src'
Making all in doc
make[2]: Entering directory '/var/lib/homebridge/bcm2835-1.71/doc'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/var/lib/homebridge/bcm2835-1.71/doc'
make[2]: Entering directory '/var/lib/homebridge/bcm2835-1.71'
make[2]: Leaving directory '/var/lib/homebridge/bcm2835-1.71'
make[1]: Leaving directory '/var/lib/homebridge/bcm2835-1.71'
pi@homebridge:/var/lib/homebridge/bcm2835-1.71 $ sudo make check
Making check in src
make[1]: Entering directory '/var/lib/homebridge/bcm2835-1.71/src'
make  test
make[2]: Entering directory '/var/lib/homebridge/bcm2835-1.71/src'
make[2]: 'test' is up to date.
make[2]: Leaving directory '/var/lib/homebridge/bcm2835-1.71/src'
make  check-TESTS
make[2]: Entering directory '/var/lib/homebridge/bcm2835-1.71/src'
make[3]: Entering directory '/var/lib/homebridge/bcm2835-1.71/src'
FAIL: test
============================================================================
Testsuite summary for bcm2835 1.71
============================================================================
# TOTAL: 1
# PASS:  0
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See src/test-suite.log
Please report to mikem@airspayce.com
============================================================================
make[3]: *** [Makefile:669: test-suite.log] Error 1
make[3]: Leaving directory '/var/lib/homebridge/bcm2835-1.71/src'
make[2]: *** [Makefile:777: check-TESTS] Error 2
make[2]: Leaving directory '/var/lib/homebridge/bcm2835-1.71/src'
make[1]: *** [Makefile:851: check-am] Error 2
make[1]: Leaving directory '/var/lib/homebridge/bcm2835-1.71/src'
make: *** [Makefile:365: check-recursive] Error 1

NorthernMan54, 2022-05-02:
humm, what is in this file `src/test-suite.log` ?

Spudaaa, 2022-05-03:
[image]

NorthernMan54, 2022-05-03:
Sounds like a permission issue with /dev/mem, try running the test again with `sudo`

Spudaaa, 2022-05-05:
Update: 
Library installed.. took another re-flash of the micro Sd and not to install the web ui until after.

See logs now not temperature readings

Log:

05/05/2022, 06:58:42] Loaded config.json with 2 accessories and 1 platforms.
[05/05/2022, 06:58:42] Loaded 0 cached accessories from cachedAccessories.
[05/05/2022, 06:58:42] ---
[05/05/2022, 06:58:44] Loaded plugin: homebridge-config-ui-x@4.43.2
[05/05/2022, 06:58:44] Registering platform 'homebridge-config-ui-x.config'
[05/05/2022, 06:58:44] ---
[05/05/2022, 06:58:50] Loaded plugin: homebridge-dht@0.1.33
[05/05/2022, 06:58:50] Registering accessory 'homebridge-dht.Dht'
[05/05/2022, 06:58:50] ---
[05/05/2022, 06:58:50] Loading 1 platforms...
[05/05/2022, 06:58:51] [Config] Initializing config platform...
[05/05/2022, 06:58:51] [Config] Running in Service Mode
[05/05/2022, 06:58:51] Loading 2 accessories...
[05/05/2022, 06:58:51] [cputemp] Initializing Dht accessory...
[05/05/2022, 06:58:51] [cputemp] Adding Accessory
[05/05/2022, 06:58:51] [cputemp] INIT: cputemp
[05/05/2022, 06:58:51] [Temp/Humidity Sensor] Initializing Dht accessory...
[05/05/2022, 06:58:51] [Temp/Humidity Sensor] Adding Accessory
[05/05/2022, 06:58:51] [Temp/Humidity Sensor] INIT: Temp/Humidity Sensor
Setup Payload:
X-HM://0023Z6QTA2Q7Q
Enter this code with your HomeKit app on your iOS device to pair with Homebridge:
                       
    ┌────────────┐     
    │ 306-61-694 │     
    └────────────┘     
                       
[05/05/2022, 06:58:51] Homebridge v1.4.1 (HAP v0.10.2) (Homebridge F7E6) is running on port 51872.
[05/05/2022, 06:58:52] [cputemp] cputemp function failed: Error: spawn cputemp ENOENT
[05/05/2022, 06:58:52] [Temp/Humidity Sensor] Error: [Error: failed to read sensor]
[05/05/2022, 06:58:52] [homebridge-dht] This plugin generated a warning from the characteristic 'Current Temperature': characteristic value expected valid finite number and received "undefined" (undefined). See https://homebridge.io/w/JtMGR for more info.

Spudaaa, 2022-05-06:
[image]

NorthernMan54, 2022-05-07:
This version should not use `[cputemp] cputemp function failed: Error: spawn cputemp ENOENT` for the DHT Temperature.  What is in your config.json ?

The error `Error: [Error: failed to read sensor]` is coming from the library [node-dht-sensor](https://github.com/momenso/node-dht-sensor).  Do you have the wiring correct ?

Spudaaa, 2022-05-08:
Wiring is exactly to your specs

Config now stands at

{
    "bridge": {
        "name": "Homebridge F7E6",
        "username": "********",
        "port": 51164,
        "pin": "*******"
    },
    "accessories": [
        {
            "accessory": "Dht",
            "name": "test",
            "gpio": "4",
            "service": "dht22",
            "refresh": "1"
        }
    ],
    "platforms": [
        {
            "name": "Config",
            "port": 8581,
            "platform": "config"
        }
    ]
}

Spudaaa, 2022-05-08:
[image]

This is the sensor used.. I have got another 2 incase of sensor failure

closed 2022-05-09 by Spudaaa