NorthernMan54/homebridge-dht #54: fakegato for history graphs

tmur11, 2022-12-29:
Is it possible to view the history data within the Eve app?

NorthernMan54, 2022-12-29:
It should, as the ability to generate the data behind the graphs is included.

[image]

tmur11, 2022-12-31:
Thanks for confirming! I had been using the other [DHT ](https://github.com/RaresAil/homebridge-temperature-sensor-dht#readme) plugin which doesn't appear to support history.

My issue is now your plugin isn't "verified" and fails to install throught homebridge-ui and installing manually gets uninstalled randomly after 20minutes

NorthernMan54, 2022-12-31:
Verified is a paper exercise, and a flag and should not impact installation.

This plugin uses a binary I complied 5 years ago and included within the plugin.  Their is possibilty that more modern RPI's don't support the binary.

When you install from the command line, what does it show ?

HBgotMeHere, 2023-01-02:
I was able to install the Plugin via the UI, but after the configuration I got this error.

[02/01/2023,
01:55:451ニニニニニニニニニニニニニニニニニニニニ
[02/01/2023, 01:55:45] ERROR LOADING PLUGIN homebridge-dht:
[02/01/2023, 01:55:45] Error: Cannot find module /build/Release/node_dht_sensor'
Require stack:
/var/lib/homebridge/node modules /homebridge-dht/node modules /node-dht-sensor /lib/ index. is /var/lib/homebridge/node modules /homebridge-dht/index. is /var/lib/homebridge/node_modules/homebridge/lib/plugin.js
- /var/lib/homebridge/node modules /homebridge/lib/pluginManager.is
/var/lib/homebridge/node modules/homebridge/lib/server. js /var/lib/homebridge/node modules/homebridge/lib/cli.js /var/lib/homebridge/node modules /homebridge/bin/homebridge
at Function. Module. resolveFilename (node: internal/modules /cis/loader: 995:15) at Function. Module. load (node: internal/modules /cis/loader: 841:27) at Module.require (node: internal/modules /cjs/loader:1061:19) at require (node: internal/modules /cis/helpers: 103:18)
at Object. <anonymous> (/var/lib/homebridge/node modules /homebridge-dht/node modules /node
-dht-sensor/lib/index. js: 1:14)
at Module. compile (node: internal/modules/cjs/loader: 1159:14)
at Object. Module. extensions.. js (node: internal/modules/cjs/loader: 1213:10) at Module. load (node: internal/modules /cis/loader: 1037: 32) at Function. Module. load (node: internal/modules /cis/loader: 878:12) at Module.require (node: internal/modules/cjs/loader: 1061:19)

After installation via the UI Console nothing showed in the Plugin tab.

NorthernMan54, 2023-01-02:
node_dht_sensor is this package - https://github.com/momenso/node-dht-sensor

And looking at the issues, they mention https://github.com/momenso/node-dht-sensor/issues/135 - bcm lib not available anymore on OS bullseye

I'm wondering if this is the case ?

https://github.com/momenso/node-dht-sensor#reference-for-building-from-source

HBgotMeHere, 2023-01-04:
I was able to fix the problem, Bit now it simply tells me failed to read Sensor. With a Python Script I am able to see values, so I think it is wired correctly.

HBgotMeHere, 2023-01-04:
Suddenly just one magic value. 

[04/01/2023, 01:41:33] [dht22] DHT Status: 0, Temperature: 21.5°C, Humidity: 65.7% [04/01/2023, 01:41:33] [dht22]
**Fakegato-timer: addData
time: [1672792893](tel:1672792893), temp: 21
.5, humidity: 65.7 } immediate: false [04/01/2023, 01:41:33] [dht22]
**Start Global Fakegato-Timer - 10min**
[04/01/2023, 01:42:35] [dht22] Error: [Error: failed to read sensor]
[04/01/2023, 01:43:35] [dht22] Error: [Error: failed to read sensor]

tmur11, 2023-01-04:
Oddly the plugin will install when done via hb-shell but then uninstalls after 30min/rebooting the pi. Install with the UI fails with the below everytime.

`343 timing reifyNode:node_modules/homebridge-dht/node_modules/moment Completed in 125594ms
344 timing reifyNode:node_modules/homebridge-dht/node_modules/googleapis Completed in 307585ms
345 timing reify:unpack Completed in 307958ms
346 timing reify:rollback:createSparse Completed in 20994ms
347 timing reify:rollback:retireShallow Completed in 2ms
348 timing command:install Completed in 406862ms
349 verbose stack Error: process terminated
349 verbose stack     at /opt/homebridge/lib/node_modules/npm/node_modules/@npmcli/arborist/lib/arborist/reify.js:199:39
349 verbose stack     at process.<anonymous> (/opt/homebridge/lib/node_modules/npm/node_modules/@npmcli/arborist/lib/signal-handling.js:53:5)
349 verbose stack     at process.emit (node:events:525:35)
350 verbose cwd /var/lib/homebridge
351 verbose Linux 5.15.76+
352 verbose node v18.12.1
353 verbose npm  v8.19.2
354 error process terminated
355 error signal SIGTERM
356 verbose exit 1
357 timing npm Completed in 412731ms
358 verbose unfinished npm timer reify 1672810996795
359 verbose code 1
360 error A complete log of this run can be found in:
360 error     /home/pi/.npm/_logs/2023-01-04T05_43_11_025Z-debug-0.log`

NorthernMan54, 2023-01-04:
What os image are you using, as I have never seen a plugin un-install itself

tmur11, 2023-01-04:
I am using the offical homebridge image downloaded from the RPI imaging tool. I am running it on a RPi B 1st gen Rev2 board

NorthernMan54, 2023-01-04:
I was chatting with some of the other plugin developers, and one comment made was that your SD card may be old/bad and causing the rollback.

HBgotMeHere, 2023-01-04:
I am using a SanDisk Ultra 16gb with an Pi Zero WH

Link for the Typ of SD-card that I use: https://www.westerndigital.com/de-de/products/memory-cards/sandisk-extreme-uhs-i-microsd#SDSQXAF-032G-GN6MA

tmur11, 2023-01-04:
Thanks @NorthernMan54 if I hadn't just flashed a brand new sandisk extreme pro 128. I suppose that doesn't mean the SD card didn't corrupt and particularly possible on the lowest end Rpi hardware. I will try a fresh flash tonight and see what happens. Wondering if i should try manually install homebridge instead of using the pre-built image? Also I'm using a [20w Anker Nano](https://www.anker.com/ca/products/a2633?variant=41698748367012) USB-C charger should be more then plenty of smooth DC juice?

NorthernMan54, 2023-01-04:
If I get some free time over the next few days, I'm to test that image an older RPI 2 that I have in storage.  I'm curios if I can recreate the 30 minute uninstall issue

NorthernMan54, 2023-01-05:
Suggestion from @ebaauw

`To confirm a suspect SD card, just do a touch x, check that file x has been created, reboot, check that x has disappeared.  Raspberry Pi OS happily writes any changes to the file system cache, and you never notice that they’re not flushed to the uSD card, until you reboot the Pi.
It’s unbelievable, until you’ve experienced it yourself.
`