NorthernMan54/homebridge-dht #55: Update plugin to support 'Current Temperature' read handler?

tomtastic, 2023-12-06:
### Current Situation

Hello,
I use this plugin to run a script to retrieve some temps (not via a DHT sensor), and though it works fine, it does throw the following errors :
```
[12/6/2023, 6:05:28 PM] [homebridge-dht] This plugin generated a warning from the
characteristic 'Current Temperature': characteristic value expected valid finite number
and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.
```
```
[12/6/2023, 6:05:16 PM] [homebridge-dht] This plugin slows down Homebridge. The
read handler for the characteristic 'Current Temperature' didn't respond at all!.
Please check that you properly call the callback! See https://homebridge.io/w/JtMGR for more info.
[12/6/2023, 6:05:38 PM] [Cupboard Ambient] Got Temperature of 32
[12/6/2023, 6:05:38 PM] [Cupboard Ambient] Got Temperature of 32
[12/6/2023, 6:05:41 PM] [Cupboard Ambient] Got Temperature of 32
```

My config is :
```
    "accessories": [
        {
            "accessory": "Dht",
            "cputemp": "/homebridge/sensors_homebridge",
            "name": "Cupboard Ambient",
            "refresh": 300,
            "service": "Temperature"
        }
```

### Proposed Change

Apply the suggestions from https://github.com/homebridge/homebridge/wiki/Characteristic-Warnings ?

NorthernMan54, 2023-12-06:
Two things are going on here, which in most cases can be ignored

1 - [12/6/2023, 6:05:28 PM] [homebridge-dht] This plugin generated a warning from the
characteristic 'Current Temperature': characteristic value expected valid finite number
and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.

The number received from the sensor / sensor driver code was likely an error message or something that wasn't a number.  Unless it is constantly repeating, it can be ignored.

2 - [12/6/2023, 6:05:16 PM] [homebridge-dht] This plugin slows down Homebridge. The
read handler for the characteristic 'Current Temperature' didn't respond at all!.
Please check that you properly call the callback! See https://homebridge.io/w/JtMGR for more info.

And here, it appears that nothing was returned from the sensor / sensor driver code.  Unless it is constantly repeating, it can be ignored.

These code be a minor coding issue or a wiring / connectivity issue with the sensor or a slightly flakey sensor.

Unless it stops working, I'm thinking it can be ignored.

tomtastic, 2023-12-06:
Ok, thank you!